roleoroleo/yi-hack-MStar #180: Unable to save configuration

ichigo86x, 2020-05-13:
After updating to 0.3.2 it is impossible to use the saved configuration. When I save the settings from the web interface, the "Saved" message appears. (Cloud disabled, RTSP Stream LOW etc.). After the restart, all configurations return to another state. (Cloud enabled, RTSP Stream HIGH etc.). Even if I change the page after saving, returning to the configuration page, the settings return to the initial state.

roleoroleo, 2020-05-13:
Try to clean the cache of the browser.

ichigo86x, 2020-05-13:
After trying to make the changes with another browser (Firefox instead of Chrome) the settings were saved correctly. Updated the page from Chrome, without cache, and the changes appeared correctly.

closed 2020-05-13 by ichigo86x