roleoroleo/yi-hack-MStar #181: rtsp streams in both qualities at the sme time don't work on 0.3.3

bonuzzz, 2020-05-14:
I think each stream needs own dedicated port anyway

roleoroleo, 2020-05-14:
It's not necessary.
You can have both stream on the same port.

bonuzzz, 2020-05-17:
I got working low only stream just right after reboot. It freezes for second each 4-5 seconds or it crashed after 14 sec of stream.  crashed without audio.

roleoroleo, 2020-05-17:
What is your model suffix?

bonuzzz, 2020-05-17:
y203c

roleoroleo, 2020-05-17:
I'm using y203c to test the fw.
What's your config?

roleoroleo, 2020-05-17:
Try this beta:
[y203c_0.3.3.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4640008/y203c_0.3.3.tar.gz)

bonuzzz, 2020-05-17:
https://gist.github.com/bonuzzz/1b6c7e968c880f89d81c0d0bb08488ce
this is my config

roleoroleo, 2020-05-17:
Apart from the cloud, it is the same configuration that I use.

bonuzzz, 2020-05-17:
on tested fw rtsp works only on high res profile without audio. highres with audio process doesn't crashed, but stream in vlc doesn't work. in low res profile without audio rrtspserver process crashed and restarted. both streams profile also doesn't work

roleoroleo, 2020-05-17:
I would be grateful if you would help me do some diagnostics.
Here another version:
[y203c_0.3.3.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4640197/y203c_0.3.3.tar.gz)

bonuzzz, 2020-05-17:
no problem. what to check exactly?

roleoroleo, 2020-05-17:
Rtsp high/low with and without audio.

bonuzzz, 2020-05-17:
nothing changes

roleoroleo, 2020-05-17:
What client are you using?

bonuzzz, 2020-05-17:
vlc on macos

roleoroleo, 2020-05-18:
Please go here:
https://github.com/roleoroleo/yi-hack-MStar/issues/183

closed 2020-05-18 by bonuzzz