roleoroleo/yi-hack-MStar #182: Set settings in UI has some problems

hagenuck1, 2020-05-14:
When setting settings mqtt can’t be saved right:
IP is Set as mqttpassword and all other fields are empty after a reboot. 
Regardless what’s set in configuration it sets the field DISABLE_CLOUD=&PASSWORD
Regardless what’s chosen in the UI. 

Also if fields are uploaded by FTP it changes them to this values after a reboot, too.

mgiako, 2020-05-16:
Same UI problems here, 
Can't save Hostname, RTSP audio settings,or any other field.
I can save the settings but  as soon I reload the page the changes are gone.
A reboot does not solve the issue.
The strange thing is that only 3 of 4 of my cams have this issue.
I have 2 9FUS and 2 BFUS. The only working is one of the 2 BFUS

I had to go back to 0.3.1 for 3 of them. What could it be?

roleoroleo, 2020-05-16:
Please, clean the cache.

mgiako, 2020-05-16:
Alresdy did that, like a hundred times.
I updated with the UI command, maybe I have to try with the sd card method?

Extreme79, 2020-05-16:
Same for me.
I lost hostnames, timezone, impossible to save and so on.
Cleaned the cache, reloaded.
Chrome browser

roleoroleo, 2020-05-16:
Something is wrong on the config files.
Try to reload a previous version and then upgrade again.

hagenuck1, 2020-05-16:
I had to re flash the image (using an older version first and then 0.3.3 because of the hash value) and afterwards I was able to configure it by deleting the cache and using private browsing on my iPhone.

mgiako, 2020-05-16:
I initially upgraded from 0.3.1 to 0.3.2 and find the issue. 
I reflashed to 0.3.1 to get a working version. 
When version 0.3.3 came out I updated again and had the same problem. I downgraded to 0.3.1 again.
@hagenuck1 Which old version did you install?

mgiako, 2020-05-16:
Ok I updated again and the problem is solved. Thanks everyone!

closed 2020-05-16 by roleoroleo