roleoroleo/yi-hack-MStar #184: Get motion detection and bay cry detection in homekit ?

mimes123, 2020-05-18:
Hi is it possible to get the motion detection in homekit and if so can someone please show me the config for it ?

and one more thing is it possible to see the last recorded video in homekit. and if so what is the config for this ?

sorry i am new in this. :)

roleoroleo, 2020-05-18:
You can configure mqtt to send messages to your broker.
And you can get the last recorded video using web services (see wiki).

mimes123, 2020-05-18:
Yes but how do I configure this in my config in homebridge ?
I can see the link in wiki but don’t know what to do with it ?

roleoroleo, 2020-05-18:
Sorry but I don't use homebridge.
Wait for answers from other users.

hairychris, 2020-06-14:
Easiest thing to do would be to add Home Assistant in the middle. You'd want to setup the MQTT plugin and then add an MQTT Binary Sensor. You can then expose this to HomeKit.

closed 2024-05-10 by github-actions[bot]