roleoroleo/yi-hack-MStar #187: Tutorial -- Yi 1080p Home 9FCN use proxychains-ng to connect your app

x571, 2020-05-25:
I got my camera from taobao (2019/11) and finally got it worked in US. sharing my tips and hopefully it helps. Thanks a lot!

1) I used y203c_0.3.3.tgz
2) use YI HOME from china app store (iOS)
3) connect to wifi and you shall hear "wifi is connected successfully ... this camera can only be used in China"
4) find your camera ip from router - yi-hack4
5) use web ip:8080 to turn off unnecessary features -- https://github.com/roleoroleo/yi-hack-MStar/issues/52#issuecomment-591847174 & enable proxychains-ng, restart
6) use putty to connect through ssh, user name - root, empty password
7) find a proxy -- http://free-proxy.cz/en/proxylist/country/CN/socks/ping/all
8) edit etc/proxychains.conf to add a proxy server. It is recommended to use socks4/5
9) test proxy by following command
proxychains4 wget -O- -q http://ipinfo.io
10) keep trying until test is passing. 
11) reboot and you shall be able to connect to your app
12) credit to this ticket -- Thank you! https://github.com/shadow-1/yi-hack-v3/issues/12

roleoroleo, 2020-05-25:
Thank you for your contribution,
If you agree, I would like to create a page on the wiki.

x571, 2020-05-25:
> Thank you for your contribution,
> If you agree, I would like to create a page on the wiki.

sure. would like to help others.

roleoroleo, 2020-05-25:
Done.
https://github.com/roleoroleo/yi-hack-MStar/wiki/CN-camera-and-proxychains-(thanks-to-@x571)

michael11123, 2020-05-27:
my 9FCN worked to step 5. but after reboot, my camera even reset. and then it can't connect wifi anymore. can you help me?

michael11123, 2020-05-28:
> > Thank you for your contribution,
> > If you agree, I would like to create a page on the wiki.
> 
> sure. would like to help others.

do you try to upload to gdrive?

bongdotcom, 2020-08-05:
I have the same issue.  Just get 9FCN and tried put most of the file in micro sd root but no lucks don't work to flash.  Even download the latest firmware, still couldn't flash.  I tried y203c_0.3.3.tgz and 0.3.5.  Tried to change the name to xxx_y20m.  Can anyone help.  Thnaks.

closed 2020-08-20 by roleoroleo