roleoroleo/yi-hack-MStar #188: 0.3.3 broke camera

radokristof, 2020-06-02:
For me after upgrading from 0.3.1 to 0.3.3 most of the features are not working.
I can't change anything in the web interface. After a restart it jumps back to the previous value.
Snapshot only loads once after a reboot and after the first picture I can't get any snapshots.
RTSP streams also don't work.

What should I try?

bonuzzz, 2020-06-02:
clean cache in web browser. like ctrl+f5 on win or cmd+r on mac

radokristof, 2020-06-02:
Tried the most obvious parts, no success. Also I use self-written scripts which gets the snapshot directly, so there is no cache involved, these also don't work.
Tried clearing the cache again, no success.

bonuzzz, 2020-06-02:
try incognito mode in browser

radokristof, 2020-06-02:
Same... Why do you think this is a client issue?
For me it really seems something is wrong with this version - but I don't see anyone else having problems like this, so maybe the update failed?!

radokristof, 2020-06-02:
Back to 0.3.1 -> everything is working again...

bonuzzz, 2020-06-02:
because I had the same problem before and there are at least two issues with the same problem with web interface. anyway 0.3.1 is most stable now. 0.3.3 has issues with rtsp and snapshots.

radokristof, 2020-06-03:
Well, I'm back on 0.3.1 and the issue is still there. It worked for a few hours so it is better than on 0.3.3 but still not the same as before... Maybe I need to wipe the sd card.

roleoroleo, 2020-06-03:
Please, send me the list of the running processes.
Login with ssh and command "ps".

What's your model suffix?

radokristof, 2020-06-03:
Model suffix: Yi 1080p Home BFUS

Processes:

    1 root      0:01 /init
    2 root      0:00 [kthreadd]
    3 root      1:21 [ksoftirqd/0]
    5 root      0:00 [kworker/0:0H]
    6 root      0:00 [kworker/u2:0]
    7 root      2:06 [rcu_preempt]
    8 root      0:00 [rcu_sched]
    9 root      0:00 [rcu_bh]
   10 root      0:00 [watchdog/0]
   11 root      0:00 [khelper]
   12 root      0:00 [writeback]
   13 root      0:00 [crypto]
   14 root      0:00 [bioset]
   15 root      0:00 [kblockd]
   16 root      0:00 [kworker/0:1]
   17 root      0:00 [kswapd0]
   18 root      0:00 [fsnotify_mark]
   31 root      0:00 [SCLDAZA_THREAD]
   32 root      0:00 [VIPDazaTask]
   36 root      0:00 [deferwq]
   39 root      0:00 /bin/ueventd
   40 root      0:01 [jffs2_gcd_mtd3]
   41 root      0:00 [jffs2_gcd_mtd2]
   76 root      0:01 [mmcqd/0]
   86 root      0:00 [cryptodev_queue]
   97 root      0:00 [spi0]
  124 root      0:00 [kworker/0:2]
  129 root      0:00 [cfg80211]
  139 root      0:00 ./log_server
  140 root      1:23 ./dispatch
  210 root      0:00 [kworker/u2:3]
  254 root      0:38 [RTW_CMD_THREAD]
  451 root      0:24 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf
  459 root      0:11 ./mp4record
  472 root      0:08 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
  481 root      0:00 telnetd
  493 root      0:00 pure-ftpd (SERVER)
  504 root      0:00 dropbear -R
  513 root      0:03 mqttv4
  532 root      0:00 udhcpc -i wlan0 -b -s /home/app/script/default.script
  619 root      0:00 ntpd -p pool.ntp.org
  626 root     21:05 h264grabber_h -r high
  627 root      4:35 rRTSPServer_h
  635 root      0:18 {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
  675 root      0:00 onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi H
  681 root      0:00 /usr/sbin/crond -c /var/spool/cron/crontabs/
 1520 root      0:00 sleep 10
 1524 root      0:00 dropbear -R
 1615 root      0:00 -sh
 1691 root      0:00 ps

roleoroleo, 2020-06-03:
y23, y25, y203c or h201c?
What's the yi firmware version?

radokristof, 2020-06-03:
y203c

4.5.0.0C_201910080934
--

roleoroleo, 2020-06-03:
Same model that I use for the tests.
The log is correct, I don't understand.

Try to dump the first bytes of the buffer:
```
hexdump -C -n 16 /dev/fshare_frame_buf
```
and check if it changes.

radokristof, 2020-06-03:
Downgrading usually works without problems?
I think the SD card might have died, do you think it can cause issues like this?
The frame buffer does not seems to change over time...

roleoroleo, 2020-06-03:
Looking better at the log, there is a problem.
rmm process (the core yi process) is absent.
Something causes the process to die.
Enable cloud and try to reset the cam using the reset hole.

radokristof, 2020-06-04:
If I reset it with the hole, I have to enter the Wi-Fi and everything again? If yes, this can only be done through the official Yi app?

roleoroleo, 2020-06-05:
Yes.
Use before the app to pair the cam and then reinstall the hack (0.3.4).

radokristof, 2020-06-05:
Thanks! I will try it soon.

radokristof, 2020-06-14:
@roleoroleo Ok, I did a reset and configured everything, wiped SD card.
The problem was still there, so it made me thinking. I'm using openHAB as my smart home controller and I started to use the IP camera addon for it. 
It uses ONVIF to fetch snapshot, and stream urls, but after it simply works with ffmpeg mainly. I don't know why, but that breaks the camera... Even turning off ONVIF won't help.
I will also contact the developer of that addon, but it seems that this is not a problem of the camera, rather that addon. Just documenting my findings here, maybe it is helpful for others and also maybe you might have some clue what can cause this....

Anyway closing this, because it seems that this is not related to the camera and the fw mainly...

radokristof, 2020-06-26:
Well even on 0.3.4 and everything I mentioned turned off, it still breaks after a few days...

roleoroleo, 2020-06-26:
What kind of break?