roleoroleo/yi-hack-MStar #189: MQTT sending only motion-stop messages

joekerna, 2020-06-03:
This is a followup issue of issue #175 

On release 0.3.3 I only receive MQTT motion-stop messages, never motion-start messages.
My steps were:

- Flash 0.3.2
- Flash 0.3.3
- Activate MQTT

My mqttv4.conf is

```
###############################################################################
#                       Configuration file for mqttv4                         #
###############################################################################

# -----------------------------------------------------------------------------
# Basic MQTT settings
# -----------------------------------------------------------------------------

MQTT_IP=192.168.178.26
MQTT_PORT=1883

MQTT_CLIENT_ID=yi-cam

MQTT_USER=yicam
MQTT_PASSWORD=12345

MQTT_PREFIX=yicam

# -----------------------------------------------------------------------------
# Set the topics where the messages will be published
# The final topic will be MQTT_PREFIX/TOPIC_MOTION
# -----------------------------------------------------------------------------

TOPIC_MOTION=motion_detection
TOPIC_MOTION_FILES=motion_files
TOPIC_BABY_CRYING=baby_crying

# -----------------------------------------------------------------------------
# Set the topics messages
# -----------------------------------------------------------------------------

MOTION_START_MSG=motion_start
MOTION_STOP_MSG=motion_stop
BABY_CRYING_MSG=crying

# -----------------------------------------------------------------------------
# Other settings
# -----------------------------------------------------------------------------

MQTT_KEEPALIVE=120
MQTT_QOS=1
MQTT_RETAIN=1
```

On my MQTT client (Mosquitto) I am subscribing to the topic `yicam/motion_detection`.
Its output is

```
mosquitto_sub -u yicam -P 12345 -t yicam/motion_detection
motion_stop
motion_stop
motion_stop
motion_stop
motion_stop
motion_stop
motion_stop
motion_stop
```

bonuzzz, 2020-06-03:
what is your full camera config? and do you have sd inserted?

joekerna, 2020-06-03:
Yes, SD card is inserted. Hasn't been removed since flashing.
motion_stop is not periodically. I've faced the camera towards a wall and I'm not getting any motion detected at all (as expected). Before the camera was facing myself and motion was detected around every 1 to 5 minutes.

Camera settings are
- camera: on
- save video when motion is detected: on
- detection sensitivity: low
- baby crying detect: off
- status led: off
- IR led: on
- rotate: off

roleoroleo, 2020-06-04:
Try the new version.
I updated mqttv4.

joekerna, 2020-06-05:
I've updated to 0.3.4 but I still get
```
mosquitto_sub -u yicam -P 12345 -t yicam/motion_detection
motion_stop
motion_stop
motion_stop
```

But `mqttv4` also states version `v0.0.3`.

```
/home/yi-hack/bin # mqttv4
Starting mqttv4 v0.0.3
key: MQTT_RETAIN | value: 1
Unrecognized config.
Trying to connect... Client yi-cam sending CONNECT
```

Is that correct?

roleoroleo, 2020-06-08:
Yes, I didn't update v0.0.3 version.

Please, try to run the app with the log as you already done here https://github.com/roleoroleo/yi-hack-MStar/issues/175#issuecomment-630994411
But ignore "GOT UNRECOGNIZED MESSAGE" and wait for a motion detection event.
You have to see something about motion.

roleoroleo, 2020-06-11:
> /home/yi-hack/bin # mqttv4
> Starting mqttv4 v0.0.3
> key: MQTT_RETAIN | value: 1
> Unrecognized config.

Please check the config file and repeat the test.
Probably you miss retain params (added in the last version).

joekerna, 2020-06-11:
```
/home/yi-hack/bin # mqttv4
Starting mqttv4 v0.0.3
Trying to connect... Client yi-cam sending CONNECT
Client yi-cam received CONNACK (0)
. connected!
Client yi-cam sending PINGREQ
Client yi-cam received PINGRESP
IPC message. Len: 11. Status: Success!
Parsing message.
4d 4f 54 49 4f 4e 5f 53 54 4f 50 
GOT MOTION STOP
CALLBACK MOTION STOP
Client yi-cam sending PUBLISH (d0, q1, r1, m1, 'yicam/motion_detection', ... (11 bytes))
Client yi-cam received PUBACK (Mid: 1)
SENDING FILES LIST
Client yi-cam sending PUBLISH (d0, q1, r1, m2, 'yicam/motion_files', ... (581 bytes))
Client yi-cam received PUBACK (Mid: 2)
Client yi-cam sending PINGREQ
Client yi-cam received PINGRESP
Client yi-cam sending PINGREQ
Client yi-cam received PINGRESP
IPC message. Len: 11. Status: Success!
Parsing message.
4d 4f 54 49 4f 4e 5f 53 54 4f 50 
GOT MOTION STOP
CALLBACK MOTION STOP
Client yi-cam sending PUBLISH (d0, q1, r1, m3, 'yicam/motion_detection', ... (11 bytes))
Client yi-cam received PUBACK (Mid: 3)
SENDING FILES LIST
Client yi-cam sending PUBLISH (d0, q1, r1, m4, 'yicam/motion_files', ... (178 bytes))
Client yi-cam received PUBACK (Mid: 4)
Client yi-cam sending PINGREQ
Client yi-cam received PINGRESP
IPC message. Len: 11. Status: Success!
Parsing message.
4d 4f 54 49 4f 4e 5f 53 54 4f 50 
GOT MOTION STOP
CALLBACK MOTION STOP
Client yi-cam sending PUBLISH (d0, q1, r1, m5, 'yicam/motion_detection', ... (11 bytes))
Client yi-cam received PUBACK (Mid: 5)
SENDING FILES LIST
Client yi-cam sending PUBLISH (d0, q1, r1, m6, 'yicam/motion_files', ... (302 bytes))
Client yi-cam received PUBACK (Mid: 6)
Client yi-cam sending PINGREQ
Client yi-cam received PINGRESP
Client yi-cam sending PINGREQ
Client yi-cam received PINGRESP
IPC message. Len: 11. Status: Success!
Parsing message.
4d 4f 54 49 4f 4e 5f 53 54 4f 50 
GOT MOTION STOP
CALLBACK MOTION STOP
Client yi-cam sending PUBLISH (d0, q1, r1, m7, 'yicam/motion_detection', ... (11 bytes))
Client yi-cam received PUBACK (Mid: 7)
SENDING FILES LIST
Client yi-cam sending PUBLISH (d0, q1, r1, m8, 'yicam/motion_files', ... (240 bytes))
Client yi-cam received PUBACK (Mid: 8)
Client yi-cam sending PINGREQ
Client yi-cam received PINGRESP

```

bonuzzz, 2020-06-11:
Is it could be other command for motion_start?
https://github.com/roleoroleo/mqttv4/blob/e4d6445ca82c6a4f71698f4f39c177db11f7ed8a/include/ipc.h#L38
Or it should show in debug log anyway?
And how do you how much is it possible to flash y25 firmware on y203c camera?

joekerna, 2020-06-11:
Should I try y203c firmware? How do I know if I have y25 or y203c?

roleoroleo, 2020-06-11:
The firmware update procedure is based on the file name.
If you don't change the name of the files, it's not possible to upgrade the wrong file.

From the log it seems that mqttv4 doesn't receive any start messages.
Try this ipc sniffer to sniff all ipc messages.
[ipc_sniff.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/4764771/ipc_sniff.tar.gz)

The MOTION_START message is:

```
#define IPC_MOTION_START        "\x01\x00\x00\x00\x02\x00\x00\x00\x7c\x00\x7c\x00"
#define IPC_MOTION_STOP         "\x01\x00\x00\x00\x02\x00\x00\x00\x7d\x00\x7d\x00"
#define IPC_BABY_CRYING         "\x04\x00\x00\x00\x02\x00\x00\x00\x02\x60\x02\x60\x00\x00\x00\x00"
```

roleoroleo, 2020-06-11:
If you want to check your model run this command:

```
/home/yi-hack # cat /home/app/init.sh | grep "update begin"
        echo "---/home/home_y203cm exist, update begin---"
        echo "---tmp/sd/home_y203cm exist, update begin---"
```

and remove the "m".

joekerna, 2020-06-11:
The y25 was correct, thanks!

IPC sniffer says:

```
/home/yi-hack # ./ipc_sniff 
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 02 00 00 00 ed 00 ed 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 02 00 00 00 ed 00 ed 00 00 00 00 00 
GOT UNRECOGNIZED MESSAGE
```

The defines in [ipc.h](https://github.com/roleoroleo/mqttv4/blob/6b0b180e19a6c3c86829a0731dfc993e41732509/include/ipc.h) are
```

#define IPC_MOTION_START        "\x4d\x4f\x54\x49\x4f\x4e\x5f\x53\x54\x41\x52\x54"
#define IPC_MOTION_STOP         "\x4d\x4f\x54\x49\x4f\x4e\x5f\x53\x54\x4f\x50"
#define IPC_BABY_CRYING         "\x42\x41\x42\x59\x5f\x43\x52\x59\x49\x4e\x47"
```

though (different from your previous comment)

roleoroleo, 2020-06-11:
It depends on which program you are debugging.
The yi subsystem sends ipc messages and ipc_multiplexer read them.
Then ipc_multiplexer sends ipc messages (on another queue) and mqttv4 read them.
If you want to debug the first messages you have to read ipc_multiplexer.h otherwise ipc.h.

About your log, start messages are not present. But if the log doesn't show them it means that yi subsytem doesn't sends. I don't understand.

For example, this is my log:
```
/tmp/sd # ./ipc_sniff
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 02 00 00 00 7c 00 7c 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
IPC message. Len: 16. Status: Success!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00
GOT UNRECOGNIZED MESSAGE
```

The 3rd is motion start.

roleoroleo, 2020-06-11:
What are your settings about motion detection on the app?

joekerna, 2020-06-12:
I can't connect to the camera using the yi-app anymore. Is that still possible? Can I revert back to the original firmware and start over?

bonuzzz, 2020-06-12:
Yi app should work if cloud enabled.
If you have dump of original firmware, so yes, otherwise it is enough to reset to defaults

roleoroleo, 2020-06-12:
> I can't connect to the camera using the yi-app anymore. Is that still possible?

If you switch off the option "Disable cloud" the app works without problems.

> Can I revert back to the original firmware and start over?

Only if you have a backup of the original firmware.
To start over you don't need the original firmware because the upgrade overwrites completely all partitions (sys and home).

joekerna, 2020-06-15:
I've reset the camera, reflashed and old version and then updated again.

Motion detection and MQTT motion start messages being sent now.

I'm sorry, I can't explain why it is working now and was not working before because I don't think I did anything different.

The issue can be closed.

A great thank you for your tremendous support!!

closed 2020-06-15 by joekerna

roleoroleo, 2020-06-15:
All's well that ends well.