roleoroleo/yi-hack-MStar #192: Poor FPS / streaming speed

gavsiu, 2020-06-06:
I have the Yi 1080p Dome BFUS with 4.6.0 and flashed yi-hack-MStar 0.3.4. It seems to be working, however I'm getting very low FPS and bitrate. Frequently at 0 fps and the RTSP stream goes blank. Bitrate is a pretty constant 40-50 KB/s.

I'm viewing the stream using tinyCam Monitor PRO on my Android phone. The video speed was fine using Yi's app on the same network, but the reason I wanted to hack the camera was so I could disable the app.

Config:
Disable cloud: On
RTSP: On
RTSP Stream: Low
RTSP Audio: None
ONVIF: Off
SSH: On
FTP: On
Telnet: Off

For comparison, my old D-Link DCS-930L gets about 15 fps and 610 KB/s on the same WiFi network.

bonuzzz, 2020-06-07:
In tinycam change h264 decoding to h264+

gavsiu, 2020-06-07:
I set video to H264 HW+ and it seems a bit better. Sometimes gets a steady 20 fps at 40-60kb/s with a delay in video (I wave my hand and it shows about 3 seconds later). Everything still goes blank once in a while.

bonuzzz, 2020-06-07:
delay is normal for all cameras and firmwares. 40-60kb/s bitrate is also normal for low resolution stream. You have the same video as in yi app without any compression etc. You could check stream stability on pc with vlc. I can't say I'm very satified with tinycam (not only with yi-hack), but I don't know better alternatives

gavsiu, 2020-06-08:
Gotta say I was expecting a bit better performance, but I guess there's nothing to do. If there's any other way to improve it, let me know please.

I switched to high resolution since the lag is the same. 20fps, avg 140 KB/s.

The stream blanking out seems to only happen when the camera boots up. Seems fine after it's been running.

Still feels like there's less lag with the JPEG stream on the D-Link.

closed 2020-06-08 by gavsiu