roleoroleo/yi-hack-MStar #195: ipc lens correction

bonuzzz, 2020-06-11:
on y203c I have two modes only
ipc codes are:
02 00 00 00 08 00 00 00 28 10 01 00 04 00 00 00 64 00 00 00

02 00 00 00 08 00 00 00 28 10 01 00 04 00 00 00 00 00 00 00

roleoroleo, 2020-06-11:
If you want to try to send these messages you can use ipc_cmd.
Try -f option to read binary code from file.

bonuzzz, 2020-06-11:
commands don't work from console

roleoroleo, 2020-06-11:
I tried it and it works.
```
/tmp/sd # echo -n -e '\x02\x00\x00\x00\x08\x00\x00\x00\x28\x10\x01\x00\x04\x00\x00\x00\x64\x00\x00\x00' > lens1.bin
/tmp/sd # echo -n -e '\x02\x00\x00\x00\x08\x00\x00\x00\x28\x10\x01\x00\x04\x00\x00\x00\x00\x00\x00\x00' > lens2.bin
/tmp/sd # ipc_cmd -f lens1.bin
/tmp/sd # ipc_cmd -f lens2.bin
/tmp/sd # ipc_cmd -f lens1.bin
```

With the 2nd ipc_cmd, the image becomes distorted.
With the 3rd ipc_cmd, the image returns normal.

bonuzzz, 2020-06-11:
great!
I used ipc_cmd wrong and with -x, without binary

closed 2020-08-20 by roleoroleo