roleoroleo/yi-hack-MStar #199: udhcpc does not send hostname

vip9, 2020-06-17:
When requesting ip address, there is no hostname send to dhcpd server on router, thus, it is hardly to find/ping/view camera.
Tried options udhcpc options to send hostname, but no luck.
Probably busybox bug?

roleoroleo, 2020-06-18:
What option are you using?

vip9, 2020-06-18:
tried -F <hostname> and -x hostname:<hostname>

roleoroleo, 2020-06-18:
Using "-x hostname:<hostname>" it works.
Captured with wireshark:

[image]

roleoroleo, 2020-06-18:
And this is the -F option:

[image]

roleoroleo, 2020-06-18:
I don't know which is the correct option.
12 or 81?

vip9, 2020-06-18:
I suppose 12, because host name is not fqdn. 
For me, running `udhcpc -i wlan0 -s /home/app/script/default.script -x hostname:`hostname`` worked.
Option 81 produced double hostname entry, `yi-hackyi-hack.lan `

vip9, 2020-08-12:
Any way to make that command permanent in wifidhcp.sh? Or somewhere else?

roleoroleo, 2020-08-13:
You could directly change the file wifidhcp.sh.

vip9, 2020-08-13:
Will it survive an upgrade?

roleoroleo, 2020-08-13:
No, sorry...

vip9, 2020-08-13:
Yup, definetely look for the hack to rewrite that file.

roleoroleo, 2020-09-13:
I added this feature:
https://github.com/roleoroleo/yi-hack-MStar/commit/8ae90a6372558f97a18473df7a61070295f86335

closed 2020-11-21 by roleoroleo