roleoroleo/yi-hack-MStar #201: FTP username/password

leeablett, 2020-06-21:
Question
I can't see to ftp to the camera are the credentials the same as the web access?

roleoroleo, 2020-06-22:
Default is root with blak password.
If you want something better read this: https://github.com/roleoroleo/yi-hack-MStar/issues/24

Rushmed, 2020-08-18:
Blank password is making trouble on FTP clients such as SolidExplorer on android which save credentials an do not allow to leave PW blank.
Would love to have a default password

roleoroleo, 2020-08-18:
Yes, Allwinner is different.
Try to change SSH password. Is the same as ftp service.

Rushmed, 2020-08-19:
> Try to change SSH password. Is the same as ftp service.

Thanks, solved the issue. Maybe hint in web UI will be useful for other users.

closed 2020-08-20 by roleoroleo