roleoroleo/yi-hack-MStar #202: BFUS serial 1080 yi cam

ilmisu, 2020-06-22:
Hi everyone, I still get problems with my 1080p Yi cam with serial "BFUS".
I could not flash the Yi hack FW, also if it is mentioned onto readme file of this repo.
Could anybody help me or tell me if maybe that kind of serial is still not supported?

thanks.

bonuzzz, 2020-06-22:
Try allwinner

ilmisu, 2020-06-22:
...for noobs?

roleoroleo, 2020-06-22:
Yes, another hack.
The model BFUS is not enough to detect the system.
You need to match fw version too.

ilmisu, 2020-06-22:
thanks, I will check out everything tomorrow..;)

LeidenSpain, 2020-07-04:
Hi all,

I have 2 BFUS that came in a pack and I am finding that if the firmware version is updated, but no hack is applied.
Right now I have one with its original (old) firmware and another updated with the hack firmware, but without being able to access it.

With another 9FUS everything went smoothly.

I see that the problem is general, but I don't know what solutions you refer to (allwinner, etc ...)

bonuzzz, 2020-07-04:
if you can't access to camera via web, it is probably your camera didn't flash hack version at all. what version of original firmware?

LeidenSpain, 2020-07-04:
One cam has the firmware version 4.5.0.0A

The hacked camera has the version 4.5.0.0C
Therefore, the new firmware has been installed

Yi 1080p Home BFUS | 4.5.0* | y203c
https://github.com/roleoroleo/yi-hack-MStar/releases/download/0.3.4/y203c_0.3.4.tgz

bonuzzz, 2020-07-04:
check again whether hacked version flash on camera or not. camera is slow flashing yellow light during process

LeidenSpain, 2020-07-04:
I am testing with version 0.2.9 and nothing happens.
That yes, I just realized that the SD I am using is 32Gb. Let's see with a 16 ...

bonuzzz, 2020-07-04:
no problem

closed 2020-08-20 by roleoroleo