roleoroleo/yi-hack-MStar #205: Home Assistant stream nosie

majkers, 2020-06-24:
Hello,
I don't know what might be the problem but using 0.3.4 version on both my cameras ie. Yi home with y203c and Yi dome with h201c I am getting noises in card displaying RTSP strem. Sometimes it looks like beloew
[image]
and the other day I have green noises. I am using Yi outdoor camera with other hack and there are no noises. What might be the problem?

roleoroleo, 2020-06-24:
The green noise is probably a frame error but, normally, it happens only when you start the rRTSPServer daemon.
1 or 2 minutes after the cam starts.

About ha, what type of card are you using?
Did you try tcp_transport option?

majkers, 2020-06-25:
I think it is not. Both cmaera are up for many days. 
In HA I am using picture-entity card with live view. Camera are configured using ONVIF integration with TCP as a transport.

bonuzzz, 2020-06-25:
do you have same image distortion in vlc?

majkers, 2020-06-26:
I do not use vlc. Only HA

bonuzzz, 2020-06-26:
You could check stream quality there to determine where the problem

roleoroleo, 2020-06-26:
Picture entity, if I'm not wrong, uses snapshot, not rtsp stream.
If you configure cam as onvif, it calls web service to get jpg image.
Probably a problem in ha + browser?

bonuzzz, 2020-06-26:
@roleo you're right. I missed @majkers uses picture entity.
I used rtsp stream in ha, but it was very long to wait to see something.. probably it uses something like pre-cache or like that

roleoroleo, 2020-06-26:
> @roleo you're right. I missed @majkers uses picture entity.
> I used rtsp stream in ha, but it was very long to wait to see something.. probably it uses something like pre-cache or like that

If you configure cam as generic or if you use an older version of ha, the picture entity card calls ffmpeg to get a frame: this means a lot of delay and a lot of cpu.
It's possible to have the stream in a picture entity adding the option "camera_view: live"
```
camera_view: live
entity: camera.kitchen
name: kitchen CAM
type: picture-entity
```

But I don't use it.

majkers, 2020-06-26:
@roleoroleo Yes. This is my exact config. Picture-entity card with camera_view: live. 
So what is the best way to display RTSP stream in HA from ONVIF cam?

roleoroleo, 2020-06-26:
I don't know what is the best way to display a stream in ha, probably it's better to use an nvr software like motion or zoneminder in the middle.
But, as @bonuzzz said above, you should check the stream with a "standard" client like vlc to understand if there are streaming problem.

bonuzzz, 2020-06-26:
@majkers I used this variant, but I have some delay before showing up video, which I don't like.

stream:
ffmpeg:
camera:
  - platform: ffmpeg
    name: Cam1
    input: -rtsp_transport tcp -i rtsp://192.168.1.121/ch0_0.h264

majkers, 2020-06-26:
@bonuzzz I had the same config before switching to ONVIF but I think it is more or less the same as transport is TCP

bonuzzz, 2020-06-26:
I thought onvif variation uses snapshot pictures from camera like mjpeg, but i'm not sure. Anyway check in vlc the stream to understand this issue is hack relative or ha

Arkezis, 2020-07-28:
I have the same issue (with Jeedom instead of Home Assistant).
2 cameras have the same issue, no problem with VLC.
Using the "low quality" stream works well (but bad quality, of course ;) )

closed 2024-05-10 by github-actions[bot]