roleoroleo/yi-hack-MStar #206: yi 1080p dome version BFCN . I open Proxychains-ng can not use

tummerry, 2020-06-25:


tummerry, 2020-06-25:
help me . I can not use in thailand[image][image]

tummerry, 2020-06-25:
no configure. How to configure?

tummerry, 2020-06-25:
Please tell me how to do

roleoroleo, 2020-06-25:
See the wiki.

tummerry, 2020-06-25:
where see wiki ?

tummerry, 2020-06-25:
IP have : or IP no have : ?
[image]
[image]

roleoroleo, 2020-06-25:
Sorry, I don't understand.

tummerry, 2020-06-25:
[image]

okok very good