roleoroleo/yi-hack-MStar #207: YI 1080 home SSH bricked

yibrick, 2020-06-27:
I did the yi-hack my MStar and everything work fine. 

I manage the cam via 8080 website and add authentication by user and pass, reboot and works fine

after that I switched off Telnet, reboot and works fine

after that I I switched off SSH and ... cant log in to cam via IP website with 8080 

restore switch did not help

how can I restart the SSH service to log in again and manage the camera through ipcamera:8080 as previously?

roleoroleo, 2020-06-28:
Disabling telnet and ssh is not enough to enable user authentication.
You have to enter username and password at the bottom of the page.

If you want to reset the yi hack, downgrade to a previous version and upgrade again.
Don't use online upgrade, use manual SD method.

yibrick, 2020-06-28:
"add the authentication by user and pass" mean I done it by the bottom of the page. everything was fine, I rebooted thecam and authentication works great on webservice and on rtsp via VLC. 

afterthat I readed that there is not so much resources so I decided to switch off services I'm not using.
at first Telnet, then reboot, login by auth to webservice and next was SSH.
after reboot as usual ipcamera:8080 website and auth box popup as always but when I'm confirm user and pass it's refreshing and again ask me for user and pass

at first I used reset button at the backside, then connected cam to WiFi as usual and try webservice:8080 - the same problem

so I decided to clear the SD card in FAT32 and made hack, same as the successful first time.
then connected cam to WiFi as usual and try webservice:8080 - the same problem

need more help

yibrick, 2020-06-28:
solved :)

I saw the the rtsp is now working without asking for user and pass. 
wondering why the webservice 8080 still pupup the box with authentication, i puted only the user name and leaved the pass blank

it works :) I was loged in again and can switch up the SSH