roleoroleo/yi-hack-MStar #208: How to add static IP address

melvinraju, 2020-06-30:
My cameras were turned off and then reset their IP addresses meaning I had to re-register them in my NVR software manually. Is there a way to give them a static IP address?

roleoroleo, 2020-06-30:
At the moment it's not possible.
I think you could create a startup.sh in the yi-hack folder of your sd and use ifconfig to change the ip address.
But I prefer to use a dhcp reservation.

bonuzzz, 2020-06-30:
create on microsd folder yi-hack and file there with name startup.sh
> #!/bin/sh
ifconfig wlan0 192.168.1.127 netmask 255.255.255.0 up
route add default gw 192.168.1.1
echo "nameserver 192.168.1.1" >> /tmp/resolv.conf

then in ssh: chmod 755 /tmp/sd/yi-hack/startup.sh and reboot

dentex, 2020-09-29:
Hello and again sorry for reopening an issue.
I tried the solution above, on hack firmware version 0.1.9, but with an Allwinner YI 1080 Home BFUS camera, with base firmware 8.2.0.0A_202007211802. The camera appears not getting always the new (fixed) IP; often it is available again at the IP assigned by the router. I tried compiling the firmware from latest master branch, successfully installed, but it behaves the same.
I also noticed that progressively it goes back to the router's IP, even if at boot it gets the fixed one. It seemed at some point that the web interface was reachable at the new IP while in the YI app there was the old one. After a few tries it was definitively on the router's IP.

dentex, 2020-09-29:
```
Nmap scan report for 192.168.1.224
Host is up (0.017s latency).
Not shown: 997 closed ports
PORT     STATE SERVICE
22/tcp   open  ssh
554/tcp  open  rtsp
8080/tcp open  http-proxy

Nmap done: 1 IP address (1 host up) scanned in 19.38 seconds
```

but `/tmp/sd/yi-hack/startup.sh` contains

```
#!/bin/sh
ifconfig wlan0 192.168.1.80 netmask 255.255.255.0 up
route add default gw 192.168.1.1
echo "nameserver 192.168.1.2" >> /tmp/resolv.conf
```

bonuzzz, 2020-09-29:
Add killall -9 udhcpc after #!/bin/sh

dentex, 2020-09-29:
I setup a fixed IP via the router itself, but I would try just to see if it works...

mimes123, 2020-09-29:
> I setup a fixed IP via the router itself, but I would try just to see if it works...
> EDIT: Yep, now it's working, just rebooted.

How do you fix the ip via the router ?

dentex, 2020-09-30:
> > I setup a fixed IP via the router itself, but I would try just to see if it works...
> > EDIT: Yep, now it's working, just rebooted.
> 
> How do you fix the ip via the router ?

I think it may depend on the router, but on mine you have to go to settings, LAN, find MAC address for the device and then open a dialog window where you can specify a fixed IP for it.

dentex, 2020-09-30:
> Add killall -9 udhcpc after #!/bin/sh

This morning the cam was reverted to router's IP ...

roleoroleo, 2020-09-30:
I think there is a process (yi process) that renew the udhcpc daemon.
When this process runs udhcpc the IP address changes...
Probably the only solution is a dhcp reservation.

dentex, 2020-09-30:
Understood, I'll try again with the router.
Thanks

bonuzzz, 2020-09-30:
Or try to rename binary or change execution rights

dentex, 2020-09-30:
> Or try to rename binary or change execution rights

What binary? udhcpc?

bonuzzz, 2020-09-30:
@dentex yes. Chmod -x udhcpc

dentex, 2020-10-01:
> @dentex yes. Chmod -x udhcpc
```
~# chmod -x /sbin/udhcpc 
chmod: /sbin/udhcpc: Read-only file system
```

bonuzzz, 2020-10-01:
@dentex chmod -x /home/app/script/wifidhcp.sh

dentex, 2020-10-01:
> @dentex chmod -x /home/app/script/wifidhcp.sh

I'm sure this one would work, but I'll stay with the static lease via the router (actually it's `pihole`, but it's the same) in order to mess with the cam no more.
Thanks nonetheless.

poprostumarek, 2020-10-11:
hello, I have this same issue with startup file. I was able to change IP with SSH command but it looks like startup file is not loading during camera initialization.  rights to startup.sh was added successfully so file should be accessible. maybe loading this startup file at the beginning is missing. FRN 8.x

bonuzzz, 2020-10-11:
@poprostumarek add to file: echo 1 > /tmp/sd/1.txt and reboot to check if script ever work

poprostumarek, 2020-10-11:
Hi, it doesn't work. there is no new file in this directory.

system.sh script from home/yi-hack/script has record:

if [ -f "/tmp/sd/yi-hack/startup.sh" ]; then
    /tmp/sd/yi-hack/startup.sh

but it doesn't seem to work

bonuzzz, 2020-10-11:
@poprostumarek Unfortunately I have no ideas. Startup.sh script worked on previous build and I haven't that something changed in current
Please, post statup.sh content, probably you have mistake or typo somwhere

poprostumarek, 2020-10-12:
```
#!/bin/sh
echo 1 > /tmp/sd/1.txt
ifconfig wlan0 192.168.0.127 netmask 255.255.255.0 up
route add default gw 192.168.0.1
echo "nameserver 192.168.0.1" >> /tmp/resolv.conf
```

rights were added. 

`chmod 755 /tmp/sd/yi-hack/startup.sh`

while using only:
```#!/bin/sh
echo 1 > /tmp/sd/1.txt
```
there is this same issue.

In fileZilla I can see correct rights and path. camera was reset to factory defaults. I have used notepad++ for .sh file

bonuzzz, 2020-10-12:
@poprostumarek just checked on mine. It works. I have y203c

bonuzzz, 2020-10-12:
@poprostumarek which capacity does microsd have?

poprostumarek, 2020-10-16:
@bonuzzz I tried different card 8GB instead of 32GB and the sam result. it was y6xxx version

bonuzzz, 2020-10-16:
@poprostumarek theoretically you could add the code to system.sh, but it's risky. if possible, much better and safer to assign ip on dhcp server.

poprostumarek, 2020-10-16:
@bonuzzz I will dig into this in my free time. Thanks for checking!

SeriousPat, 2022-02-06:
> I think there is a process (yi process) that renew the udhcpc daemon. When this process runs udhcpc the IP address changes... Probably the only solution is a dhcp reservation.

even with dhcp ip reservation. the cam gets an wrongip:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/347