roleoroleo/yi-hack-MStar #209: Identify Model

phrogg, 2020-07-06:
I know this is most certainly an already answered question but, how do I identify my camera Model I found several numbers on the box as well as the camera.

**Model**:YYS.2016
I also found an **FCC ID** and an **IC**.

I really wanted to find this out myself, but I read through most of the issues here as well as the wiki as well as the readme, but didn't found that info. Or is it only possible by dumping the old fw like its said in the wiki? Is it therefore not possible to do this without soldering?

Many thanks!

bonuzzz, 2020-07-06:
on back of your camera below qr code you wiil find serial number. first four symbols mean model version. also check firmware version in yi app on phone.

phrogg, 2020-07-06:
> on back of your camera below qr code you wiil find serial number. first four symbols mean model version. also check firmware version in yi app on phone.

Lol, I thought it must be in an more or less obvious place. Thanks so much ;)

closed 2020-07-06 by phrogg