roleoroleo/yi-hack-MStar #210: Wifi password error 9FUS

stclaus77, 2020-07-06:
Hi! Firmware already flashed successfully!
Camera YYS.2016 v1.1 / 9FUS / y203c
But when I try to connect to wifi, always camera says "Wifi password error" and it repeats and repeats
I checked, Wifi is 2,4Ghz, no black/white lists. Even when I set no password on Wifi!
Don't understand what's wrong.

bonuzzz, 2020-07-06:
these firmware modifications don't involve on wifi stuff. have you tried already to reset camera and set wifi settings through qr code in yi app?

stclaus77, 2020-07-07:
I tried to reset camera many times. 
In the yi I scan QR on camera, fill password, camera scans QR code from phone, and after about 1 minutes says "Wifi password error"
I tried it on different devices : Android 8,9,10, Iphone, PC. Used different Wifi points: created hotspot on phones, used wifi routers. May I do anything wrong?

bonuzzz, 2020-07-07:
I have no idea. Probably wifi password has unsupported characters.

stclaus77, 2020-07-07:
But password only digits, and problem when I set no password on wifi

stclaus77, 2020-07-11:
Just watch log. I see that camera successful sqan QR code, ssid and pass correct! But conncetion process crashes in the end.

`
[
][1/1/0:1:25:465]: dispatch.c(main-5294) hw_type(2)
[
][1/1/0:1:28:309]: dispatch.c(main-5354) open_ptz fail[
][1/1/0:1:28:328]: dispatch.c(get_config-1239) got sn(ePlzXg8SaXYtdHBlz3DCyFk6Q7jwNM6r)[
][1/1/0:1:28:328]: dispatch.c(get_config-1240) got pwd()[
][1/1/0:1:28:328]: dispatch.c(get_config-1241) got ssid()[
][1/1/0:1:28:328]: dispatch.c(get_config-1242) got tnp_init_string()[
][1/1/0:1:28:328]: dispatch.c(get_config-1250) reset because no wifi config[
][1/1/0:1:28:328]: dispatch.c(send_saveconfig_msg-426) send_saveconfig_msg msg snd success[
][1/1/0:1:28:329]: dispatch.c(get_config-1297) get_config save conf
[
][1/1/0:1:28:329]: dispatch.c(choose_server-650) in choose_server, region_id = 17, api_server = https://api.us.xiaoyi.com, log_server = http://log.us.xiaoyi.com, sname = familymonitor-y203c, dlproto = mius
[
][1/1/0:1:28:329]: dispatch.c(choose_server-650) in choose_server, region_id = 16, api_server = https://api.eu.xiaoyi.com, log_server = http://log.eu.xiaoyi.com, sname = familymonitor-y203c, dlproto = mieu
[
][1/1/0:1:28:338]: dispatch.c(main-5438) init ok, cost time(2871 ms)[
][1/1/0:1:28:339]: dispatch.c(save_config-1020) got pwd()()[
][1/1/0:1:28:339]: dispatch.c(save_config-1021) got ssid()()[
][1/1/0:1:28:339]: dispatch.c(save_config-1022) config save finish 1[
][1/1/0:1:28:534]: dispatch.c(p_worker-3259) save_config[
][1/19/6:41:36:97]: dispatch.c(p_worker-3349) DISPATCH_SET_DEFAULT_TIME 1453185696[
][1/19/6:41:38:506]: rmm.c(msg_proc-4627) pid[490][
][1/19/6:41:39:544]: rmm.c(init_font-972) font:/home/base/iso8859-1-24x48.bin
[
][1/19/6:41:39:545]: rmm.c(init_font-984) filesize=36864
[
][1/19/6:41:39:545]: rmm.c(init_font-1004) init font /home/base/iso8859-1-24x48.bin ptr size(4)[
][1/19/6:41:39:545]: rmm.c(init_font-972) font:/home/base/iso8859-1-8x16.bin
[
][1/19/6:41:39:545]: rmm.c(init_font-984) filesize=4096
[
][1/19/6:41:39:545]: rmm.c(init_font-1004) init font /home/base/iso8859-1-8x16.bin ptr size(4)[
][1/19/6:41:42:420]: cloud.c(sys_init-5533) open share mem ok
[
][1/19/6:41:42:655]: p2p_tnp.c(main-6660) fshare_open ok[
][1/19/6:41:42:773]: p2p_tnp.c(p2p_set_tnp_init_status-2112) p2p_set_tnp_init_status 1 send_msg ok!
[
][1/19/6:41:42:811]: oss.c(main-2963) oss stream = ./oss, venc chn = 0x400[
][1/19/6:41:42:908]: oss.c(main-2982) fshare_open ok[
][1/19/6:41:43:57]: dispatch.c(do_mq_process-5109) invalid msg 0x1002
[
][1/19/6:41:43:148]: rmm.c(send_wait_msg-619) msg snd success[
][1/19/6:41:43:362]: watch_process.c(get_watch_info-116) check_interval=10
[
][1/19/6:41:43:449]: watch_process.c(get_watch_info-134) process=dispatch;cmd=cd /home/app;./dispatch &;
[
][1/19/6:41:43:450]: watch_process.c(get_watch_info-134) process=cloud;cmd=cd /home/app;./cloud &;
[
][1/19/6:41:43:450]: watch_process.c(get_watch_info-134) process=rmm;cmd=reboot;
[
][1/19/6:41:43:450]: watch_process.c(get_watch_info-134) process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
[
][1/19/6:41:43:450]: watch_process.c(get_watch_info-134) process=mp4record;cmd=cd /home/app;./mp4record &;
[
][1/19/6:41:43:450]: watch_process.c(get_watch_info-134) process=arp_test;cmd=cd /home/app;./arp_test &;
[
][1/19/6:41:43:450]: watch_process.c(get_watch_info-134) process=oss;cmd=cd /home/app;./oss &;
[
][1/19/6:41:44:872]: rmm.c(msg_proc-4667) got RMM_SPEAK_WAIT[
][1/19/6:41:53:365]: watch_process.c(check_watch_info-171) arp_test crashed![
][1/19/6:42:0:689]: dispatch.c(do_mq_process-5109) invalid msg 0x1002
[
][1/19/6:42:0:792]: rmm.c(send_wait_msg-619) msg snd success[
][1/19/6:42:2:158]: rmm.c(msg_proc-4667) got RMM_SPEAK_WAIT[
][1/19/6:42:10:217]: rmm.c(zbar_proc-4518) decoded QR-Code symbol "b=EUcBNVTmHLJE7X66&s=WlRF&p=CAx9dGJbWQt4ZQ=="
[
[... 24 lines cut ...]
][1/19/6:42:11:370]: dispatch.c(do_monitor_wifi-1711) wifi disconnected, now reconnect wifi
[
][1/19/6:42:12:604]: rmm.c(msg_proc-4752) got RMM_SPEAK_SCAN_OK[
][1/19/6:42:13:395]: watch_process.c(check_watch_info-171) cloud crashed![
][1/19/6:42:13:488]: watch_process.c(check_watch_info-171) p2p_tnp crashed![
][1/19/6:42:14:573]: rmm.c(msg_proc-4687) got RMM_SPEAK_CONNECTTING[
][1/19/6:42:23:396]: watch_process.c(check_watch_info-171) cloud crashed![
][1/19/6:42:23:461]: watch_process.c(check_watch_info-171) p2p_tnp crashed![
][1/19/6:42:33:396]: watch_process.c(check_watch_info-171) cloud crashed![
][1/19/6:42:33:396]: watch_process.c(check_watch_info-171) p2p_tnp crashed![
][1/19/6:42:36:382]: dispatch.c(p_worker-3375) rcd may crashed
[
][1/19/6:42:43:397]: watch_process.c(check_watch_info-171) cloud crashed![
][1/19/6:42:43:460]: watch_process.c(check_watch_info-171) p2p_tnp crashed![
][1/19/6:42:53:400]: watch_process.c(check_watch_info-171) cloud crashed![
][1/19/6:42:53:400]: watch_process.c(check_watch_info-171) p2p_tnp crashed![
][1/19/6:43:3:400]: watch_process.c(check_watch_info-171) cloud crashed![
][1/19/6:43:3:401]: watch_process.c(check_watch_info-171) p2p_tnp crashed![
][1/19/6:43:13:402]: watch_process.c(check_watch_info-171) cloud crashed![
][1/19/6:43:13:402]: watch_process.c(check_watch_info-171) p2p_tnp crashed![
][1/19/6:43:19:15]: dispatch.c(do_monitor_wifi-1674) timeout reset wifi
[
][1/19/6:43:19:117]: dispatch.c(do_monitor_wifi-1677) wifi connect timeout

`

bonuzzz, 2020-07-11:
I can't help you. probably @roleo knows better

roleoroleo, 2020-07-12:
Did you hacked the cam before connecting it to the wifi?

stclaus77, 2020-07-12:
At first time camera connected to wifi successfully, but when i power off it, then camera didn't conncet to wifi.
Then i flashed your yi-hack-mstar.

roleoroleo, 2020-07-12:
Maybe a hardware problem?

elmoro7, 2020-07-13:
Had a similar problem last week. After changing Router/Wifi  passphrase, I reset the camera. When I tried to connect using the app to configure the new Wifi, it would constantly time out after, the QR code got accepted and "connected to Wifi" was heard.  At this point I can connect to the yi-hack-mstar webserver and 'get access to the cam' . So it meant the camera was connected correctly to the wifi, but it failed to pair with the app, so a restart meant the camera was reset again.

How I solved this ( after many tries/resets/different routers), simply downgraded to earlier version (0.3.3 in my case), configured the wifi/paired the app, and then updated to 0.3.4

cedricous, 2020-07-18:
Hello, I have the same issue with a 9BFUS camera (yi home 1080p AI+), but me for it was at the begining with thé original firmware, impossible to connect Always the message "wifi password error", i check thé log and the password and SSID are good, and thé Wifi is 2.4ghz... So as it was not working with thé original i flash it several Times with Mstar ans Allwiner but nothing work always the same issue... Is it an error to flash a caméra which was never connected to the network ? I try to remove all hack firmware and put the original but the issue of connections to wifi still remains. Any help will be higly appreciate. Thx you.

roleoroleo, 2020-07-19:
If the cam doesn't connect with the original fw, it will not work with the hack.
Try another ap, try to get closer to the ap, try to change wifi channel...

closed 2020-11-21 by roleoroleo