roleoroleo/yi-hack-MStar #211: Can't change Configurations parameters

theos10, 2020-07-12:
On configuration page, i can't change hostname, timezone, or other parameters.
I change it, save, and reboot, but the parameters not change, the hostname stay yi-hack.
On Camera Setting web page, the setting work fine.
My camera version version is  YI Dome 1080
Firmware Version | 0.3.3
Base Version | 4.6.0.0A_201908271549
Model Suffix | h201c

I try firmware Version 0.3.4, but i can't pairing my camera.
So I downgrade to 0.3.3.
Thanks for your help.

roleoroleo, 2020-07-12:
Clean the cache.

theos10, 2020-07-12:
I do the change in ssh /home/yi-hack/etc/system.conf , /etc/hostname and /etc/TZ
in the system.conf my password is setting in DISABLE_CLOUD variable.
I change it manualy.

roleoroleo, 2020-07-12:
Try to restore factory using Maintenance page.

theos10, 2020-07-12:
I do it, but same problème

roleoroleo, 2020-07-12:
You have to clean the cache of the browser.
After that restore factory.

Sasso0101, 2020-07-24:
Had the same problem, just needed to clear the cache as @roleoroleo said and everything worked fine. Thanks! :)

theos10, 2020-07-25:
Hi, With the new firmware 0.3.5, it work fine.

feiticeir0, 2020-08-15:
Hi ! 
I'm having the same problem. Already clear the browser cache, restored factory defaults, rebooted the camera, filled in the configurations, saved, rebooted the camera and everything blank... 
Everytime I reboot the camera, everything is blank...