roleoroleo/yi-hack-MStar #212: [Feature Request] Use MQTT to enable or disable the camera

galadril, 2020-07-15:
Would it be possible to receive input events on the camera to enable or disable the camera? Or enable or disable for example motion detection, or baby crying detection?

Then we could use MQTT (within our home automation system) to automate the on/off and detection settings.?

roleoroleo, 2020-07-15:
What do you mean with "input events on the camera"?
Currently you can't change the behavior of the cam with MQTT. The cam doesn't subscribe topics.
But there are a lot of web services that you could integrate in your home automation solution.
See the wiki.

galadril, 2020-07-15:
ok, true.. webservices is a solution for my issue.. thanks for the firmware

closed 2020-07-15 by galadril

galadril, 2020-07-16:
@roleoroleo I've created an plugin for Domoticz integration.
https://github.com/galadril/Domoticz-Yi-Hack-Plugin

That works with the AllWinner and MStar (because they use the same kind of webservices).
Would you like to add that to the wiki?

roleoroleo, 2020-07-16:
Yes, thank you for your contribution.