roleoroleo/yi-hack-MStar #213: Who creates /tmp/wpa_supplicant.conf?

PieVo, 2020-07-15:
Question here:
I would like to make the (y203c) camera "portable" -> I have an old android tablet with wifi hotspot enabled and want to make the camera use that wifi connection.

Now, adding a network would be easy as adding it to wpa_supplicant.conf. However, this file is located in /tmp and thus is created during boot.

Does anybody know who creates this file? And, where is my wifi password stored? (Would be pretty bad if that came from the "cloud")

bonuzzz, 2020-07-15:
good question. I had same idea some time ago. wpa_supplicant.conf is not problem here because you can set path to config file at /home/app/script/wificonnect.sh and even locate to sd card.
But the real problem is how to skip wizard after reset.

PieVo, 2020-07-15:
So a custom wpa_supplication is indeed possible.. Either changing the wificonnect.sh script or concatenate an additional network to the one in /tmp and call "wpa_cli reconfigure".

PieVo, 2020-07-15:
I've compiled "iw" tools for the camera. To play around and to get some more info from the wireless controller.
[iw.zip](https://github.com/roleoroleo/yi-hack-MStar/files/4927644/iw.zip)

        Supported interface modes:
                 * IBSS
                 * managed
                 * AP

It supports AP mode, so running an accesspoint would also be possible. That's in practice easier than connecting to a hostpot from the camera...

bonuzzz, 2020-07-15:
/home/app/localbin/wifi_conf generates /tmp/wpa_supplicant.conf

roleoroleo, 2020-07-16:
Username and password are store (I think) in mtd5 as a raw partition.
Then is copied to /tmp/mmapinfo as a binary file (probably a direct dump of a struct).
To change it we have to write the raw partition but I don't know if it's possible.
Maybe there are crc, checksum or other data dependencies. I don't know.

Probably it's better to overwrite wpa_supplicant.conf and restart connection.

bonuzzz, 2020-07-16:
@roleoroleo do you know what process related on first run wizard? Maybe some prop sets after execution. It would be great if we find method to skip wizard

roleoroleo, 2020-07-16:
No, I don't know.
Probably rmm...

bonuzzz, 2020-07-16:
seems i'm very lucky today. i broke camera when trying to catch log from uart.  now i have only IPL gd15622▒▒▒▒▒
IPL gd15622▒▒▒▒▒▒
IPL gd156   ▒▒
 there

bonuzzz, 2020-07-16:
I've ordered one more camera. I doubt the current one will revive. I very hope it would be y203c revision and not allwinner.
Anyway I think dispatch binary starts wizard but I don't know condition. If it it checks that raw partititon, it will be very bad

PieVo, 2020-07-16:
Commonly, that kind of mumbo jumbo on the uart is a rom loader that accepts
binaries over serial (x modem).

Everything can be revived, but sometimes it takes some time and knowledge..

Looks like your bootloader is erased?

bonuzzz, 2020-07-16:
yes, i think so.  I think it was short circuit. I didn't want to solder and just leaned cable to pins. Probably I touched something when catching the log

roleoroleo, 2020-07-16:
You could try to remove it from the pcb.
And to reprogram with a low cost spi programmer.
Ftdi or parallel.

bonuzzz, 2020-07-16:
@roleoroleo I don;t have spi programmer and to be honest I don't know what exactly model I need to buy.

roleoroleo, 2020-07-17:
Try to read this old post:
https://www.ilpuntotecnico.com/forum/index.php/topic,14289.msg188607.html#msg188607
http://rayer.g6.cz/elektro/spipgm.htm

You need an old pc with a parallel port and some components.

bonuzzz, 2020-08-01:
I received programmator but couldn't determine chip type

closed 2020-11-21 by roleoroleo