roleoroleo/yi-hack-MStar #215: CPU Load very high

n3roGit, 2020-07-19:
i have flashed a BFUSY. but when i enable the rtsp the cpu load is very high. is that normal or any ideas to reduce it?

roleoroleo, 2020-07-19:
How much?
Post a top command.

n3roGit, 2020-07-20:
@roleoroleo 
[image]

n3roGit, 2020-07-20:
I have disabled nearly all stuff.

Only 
mqtt
Rtsp video low
Cloud disabled 
Is enabled. 

No motion detection record or onfiv

roleoroleo, 2020-07-20:
Please: login with ssh, run top and share the screen

n3roGit, 2020-07-20:
The problem is that the web interface and ssh is often so slow that I can hardly connect to it.  The rtsp stream also fails after a few hours.  Then only a restart of the cam helps.

I just restarted the cam and try to figure it out.

roleoroleo, 2020-07-20:
Something is wrong.

n3roGit, 2020-07-20:
Yes :D any ideas to try?

roleoroleo, 2020-07-20:
I need some logs.
Try to run top command.

n3roGit, 2020-07-21:
[image]

roleoroleo, 2020-07-21:
rmm, normally it's about 30%.
But it depends on the active services: motion detection, baby crying, face detection, etc...

n3roGit, 2020-07-22:
Everything of this services is of. After some runtime the cam doesn't give me the rtsp stream.
The web interface stucks and it's not possible to login via ssh or telnet

roleoroleo, 2020-07-22:
Very strange.
Try to reset the cam.

- Restore yi-hack default.
- Reset the cam with the button.
- Apply again the hack.

n3roGit, 2020-07-24:
Hey =)
i have ordered another cam and i have the same.
The load is high and after some time the rtsp stream failes.

The log in Motioneye give me this info "netcam_rtsp_open_context: Normal resolution: Camera image size is invalid"

n3roGit, 2020-08-24:
Now it's working for more than 24h without problems. But I have to allow all traffic. Only http and https is not enaugh.

I hope you could fix this bug :)

roleoroleo, 2020-08-24:
I'm sorry but this problem is out of my scope.
It's part of the original firmware and I can't fix it.

Maybe we could try to kill the process that run out of memory.
Are you able to check which program causes the problem?
I think a ps command is enough. Repeated every few hours.

n3roGit, 2020-08-25:
how about a script that writes a log to the SD every minute? You could even include this as debugging directly into the firmware.

Then I can upload this one to you

n3roGit, 2020-08-28:
Unfortunately I was happy too early. The error appears again :(

roleoroleo, 2020-08-28:
Create the file /tmp/sd/yi-hack/startup.sh
with this content
```
#!/bin/sh

while true; do
    echo "########################################" >> /tmp/sd/top.log
    date >> /tmp/sd/top.log
    top -b -n 2 -d 1 >> /tmp/sd/top.log
    sleep 3600
done
```

Share the log with me in a few days.

fabiosci, 2020-08-28:
> Unfortunately I was happy too early. The error appears again :(

did you check if the cam is still connected to wifi when the issue occurs? in my case it is always not connected to the wlan

n3roGit, 2020-08-31:
@fabiosci 
The cam is still online when the RTSP stream is not working anymore. I restart the cam with a CGI command to get around the problem temporarily.

The PTZ of the GUI is still working when RTSP is not working.

I had to create the folder for the log. Since I don't know if the script is now executed automatically on startup, I executed it with startip.sh &. Let's wait and see what happens here.

roleoroleo, 2020-08-31:
> @roleoroleo I see the CPU between 30 and 40%. I copied your script to add further details on this issue. Just an information: in tmp/sd there weren't a folder "yi-hack" so i created it. is it normal?

Yes, it's normal.
It's useful for additional functionality.

fabiosci, 2020-08-31:
> @fabiosci
> The cam is still online when the RTSP stream is not working anymore. I restart the cam with a CGI command to get around the problem temporarily.
>
> The PTZ of the GUI is still working when [...]

if your cam appears online your behaviour is different. from my router i see it offline. A news is that this happens also with "Disable cloud" OFF and internet connection allowed. I'm not sure that our issues are related...

n3roGit, 2020-09-01:
Last connection lost on 01.09.2020-08:46

[image]

`########################################
Fri Apr 17 22:21:23 CEST 2020
Mem: 59448K used, 1408K free, 1928K shrd, 7648K buff, 10408K cached
CPU: 36.3% usr 36.3% sys  0.0% nic 27.2% idle  0.0% io  0.0% irq  0.0% sirq
Load average: 2.08 1.38 0.62 3/88 801
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
  558     1 root     S    70204114.8   0 45.3 ./rmm
    7     2 root     SW       0  0.0   0  9.0 [rcu_preempt]
  138     1 root     S     4292  7.0   0  0.0 ./dispatch
 1248     1 root     S     3620  5.9   0  0.0 rRTSPServer
 1107     1 root     S     3584  5.8   0  0.0 mqttv4
  567     1 root     S     2256  3.6   0  0.0 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
  789   787 root     S     2068  3.3   0  0.0 grep h264grabber_l
 3339  3201 root     S     2064  3.3   0  0.0 -sh
  788   787 root     S     2064  3.3   0  0.0 top -b -n 2 -d 1
  790   787 root     S     2064  3.3   0  0.0 grep -v grep
 1252     1 root     S     2060  3.3   0  0.0 {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
 1068     1 root     S     2060  3.3   0  0.0 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
  796   794 root     R     2060  3.3   0  0.0 top -b -n 2 -d 1
  629     1 root     S     2060  3.3   0  0.0 udhcpc -i wlan0 -b -s /home/app/script/default.script
  787  1252 root     S     2060  3.3   0  0.0 {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
  791   787 root     S     2060  3.3   0  0.0 tail -n 1
  793   787 root     S     2060  3.3   0  0.0 awk {print $8}
  794  3339 root     S     2060  3.3   0  0.0 {startup.sh} /bin/sh ./startup.sh
  800   799 root     R     2060  3.3   0  0.0 [modprobe]
 1210     1 root     S     2060  3.3   0  0.0 ntpd -p 192.168.2.1
 3201  1083 root     S     2012  3.2   0  0.0 dropbear -R
 1083     1 root     S     1728  2.8   0  0.0 dropbear -R
  137     1 root     S     1512  2.4   0  0.0 ./log_server
 1088     1 root     S     1316  2.1   0  0.0 ipc_multiplexer
   39     1 root     S      408  0.6   0  0.0 /bin/ueventd
    1     0 root     S      384  0.6   0  0.0 /init
   40     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd3]
    3     2 root     SW       0  0.0   0  0.0 [ksoftirqd/0]
  354     2 root     SW       0  0.0   0  0.0 [RTW_CMD_THREAD]
   32     2 root     SW       0  0.0   0  0.0 [VIPDazaTask]
   79     2 root     SW       0  0.0   0  0.0 [mmcqd/0]
  592     2 root     SW       0  0.0   0  0.0 [kworker/u2:3]
   16     2 root     SW       0  0.0   0  0.0 [kworker/0:1]
   41     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd2]
   17     2 root     SW       0  0.0   0  0.0 [kswapd0]
    2     0 root     SW       0  0.0   0  0.0 [kthreadd]
    5     2 root     SW<      0  0.0   0  0.0 [kworker/0:0H]
    6     2 root     SW       0  0.0   0  0.0 [kworker/u2:0]
    8     2 root     SW       0  0.0   0  0.0 [rcu_sched]
    9     2 root     SW       0  0.0   0  0.0 [rcu_bh]
   10     2 root     SW       0  0.0   0  0.0 [watchdog/0]
   11     2 root     SW<      0  0.0   0  0.0 [khelper]
   12     2 root     SW<      0  0.0   0  0.0 [writeback]
   13     2 root     SW<      0  0.0   0  0.0 [crypto]
   14     2 root     SW<      0  0.0   0  0.0 [bioset]
   15     2 root     SW<      0  0.0   0  0.0 [kblockd]
   18     2 root     SW       0  0.0   0  0.0 [fsnotify_mark]
   31     2 root     SW<      0  0.0   0  0.0 [SCLDAZA_THREAD]
   36     2 root     SW<      0  0.0   0  0.0 [deferwq]
   89     2 root     SW<      0  0.0   0  0.0 [cryptodev_queue]
  100     2 root     SW       0  0.0   0  0.0 [spi0]
  121     2 root     SW       0  0.0   0  0.0 [kworker/0:2]
  132     2 root     SW<      0  0.0   0  0.0 [cfg80211]
  799   592 root     SW       0  0.0   0  0.0 [kworker/u2:3]
Mem: 59420K used, 1436K free, 1928K shrd, 7648K buff, 10408K cached
CPU: 45.0% usr 17.6% sys  0.0% nic 37.2% idle  0.0% io  0.0% irq  0.0% sirq
Load average: 2.08 1.38 0.62 1/87 808
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
  558     1 root     S    70204114.8   0 52.9 ./rmm
  796   794 root     R     2068  3.3   0  1.9 top -b -n 2 -d 1
[... 2980 lines cut ...]
 1247   175 root     S     2060  3.3   0  0.0 {startup.sh} /bin/sh /tmp/sd/yi-hack/startup.sh
  175   174 root     S     2060  3.3   0  0.0 {system.sh} /bin/sh /home/yi-hack/script/system.sh
  174    42 root     S     2060  3.3   0  0.0 {yi-hack} /bin/sh /etc/init.d/yi-hack
  659     1 root     S     2060  3.3   0  0.0 udhcpc -i wlan0 -b -s /home/app/script/default.script
 2728  1229 root     S     2060  3.3   0  0.0 {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
 2733  2728 root     S     2060  3.3   0  0.0 tail -n 1
 2735  2728 root     S     2060  3.3   0  0.0 awk {print $8}
 1050     1 root     S     1728  2.8   0  0.0 dropbear -R
  137     1 root     S     1512  2.4   0  0.0 ./log_server
 1055   175 root     S     1316  2.1   0  0.0 ipc_multiplexer
   39     1 root     S      408  0.6   0  0.0 /bin/ueventd
    1     0 root     S      384  0.6   0  0.0 /init
  351     2 root     SW       0  0.0   0  0.0 [RTW_CMD_THREAD]
  404     2 root     SW       0  0.0   0  0.0 [kworker/u2:1]
    6     2 root     SW       0  0.0   0  0.0 [kworker/u2:0]
   32     2 root     SW       0  0.0   0  0.0 [VIPDazaTask]
   10     2 root     SW       0  0.0   0  0.0 [watchdog/0]
   40     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd3]
   16     2 root     SW       0  0.0   0  0.0 [kworker/0:1]
   79     2 root     SW       0  0.0   0  0.0 [mmcqd/0]
   17     2 root     SW       0  0.0   0  0.0 [kswapd0]
   41     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd2]
    2     0 root     SW       0  0.0   0  0.0 [kthreadd]
    5     2 root     SW<      0  0.0   0  0.0 [kworker/0:0H]
    8     2 root     SW       0  0.0   0  0.0 [rcu_sched]

roleoroleo, 2020-09-01:
The log doesn't show strange behaviors.
The free memory is low but doesn't decrease.

I would like to know if the problem is only rtsp daemon.
When the stream crashes:
- Check if rRTSPServer is running
- Check if wd_rtsp.sh is running (it's a watchdog for the rtsp process)
- Try to restart only rRTSPServer (using web service or ssh)

n3roGit, 2020-09-01:
I will gladly check that.
Apparently only the stream seems to be affected.
Access to gie GUI/SSH is possible. PTZ and restarting via CLI also works.

n3roGit, 2020-09-02:
All Stuff from ps
`/home/yi-hack # ps
PID   USER     TIME  COMMAND
    1 root      0:02 /init
    2 root      0:00 [kthreadd]
    3 root      3:20 [ksoftirqd/0]
    5 root      0:00 [kworker/0:0H]
    6 root      0:00 [kworker/u2:0]
    7 root      5:57 [rcu_preempt]
    8 root      0:00 [rcu_sched]
    9 root      0:00 [rcu_bh]
   10 root      0:01 [watchdog/0]
   11 root      0:00 [khelper]
   12 root      0:00 [writeback]
   13 root      0:00 [crypto]
   14 root      0:00 [bioset]
   15 root      0:00 [kblockd]
   16 root      0:00 [kworker/0:1]
   17 root      0:00 [kswapd0]
   18 root      0:00 [fsnotify_mark]
   31 root      0:00 [SCLDAZA_THREAD]
   32 root      0:04 [VIPDazaTask]
   36 root      0:00 [deferwq]
   39 root      0:00 /bin/ueventd
   40 root      0:01 [jffs2_gcd_mtd3]
   41 root      0:00 [jffs2_gcd_mtd2]
   79 root      0:00 [mmcqd/0]
   89 root      0:00 [cryptodev_queue]
  100 root      0:00 [spi0]
  132 root      0:00 [cfg80211]
  137 root      0:02 ./log_server
  138 root      3:52 ./dispatch
  552 root      9h42 ./rmm
  586 root      0:13 [kworker/u2:3]
  816 root      0:00 [kworker/0:0]
  844 root      1:37 [RTW_CMD_THREAD]
  853 root      0:53 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf
  883 root      0:00 udhcpc -i wlan0 -b -s /home/app/script/default.script
 1047 root      0:00 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
 1062 root      0:00 dropbear -R
 1067 root      0:00 ipc_multiplexer
 1079 root      0:09 mqttv4
 1109 root      0:00 ntpd -p 192.168.2.1
 1146 root      0:26 {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
 2985 root      0:00 rRTSPServer
 3644 root      0:00 dropbear -R
 3693 root      0:00 -sh
 3803 root      0:00 {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
 3804 root      0:00 top -b -n 2 -d 1
 3805 root      0:00 grep h264grabber_l
 3806 root      0:00 grep -v grep
 3807 root      0:00 tail -n 1
 3808 root      0:00 awk {print $8}
 3815 root      0:00 ps
`
The Server is running " 2985 root      0:00 rRTSPServer"

`/home/yi-hack # netstat -an
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        6      0 0.0.0.0:554             0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
tcp       92      0 192.168.2.65:554        192.168.1.40:46122      CLOSE_WAIT
tcp        0    272 192.168.2.65:22         192.168.1.69:55382      ESTABLISHED
tcp       92      0 192.168.2.65:554        192.168.1.40:46004      CLOSE_WAIT
tcp       92      0 192.168.2.65:554        192.168.1.40:46072      CLOSE_WAIT
tcp        0      0 192.168.2.65:35990      192.168.1.40:1884       ESTABLISHED
tcp       92      0 192.168.2.65:554        192.168.1.40:46178      CLOSE_WAIT
tcp       92      0 192.168.2.65:554        192.168.1.40:46356      CLOSE_WAIT
[... 93 lines cut ...]
tcp       92      0 192.168.2.65:554        192.168.1.40:49408      CLOSE_WAIT
tcp       91      0 192.168.2.65:554        192.168.1.40:49686      ESTABLISHED
tcp        0    832 192.168.2.65:22         192.168.1.69:55382      ESTABLISHED
tcp        0      0 192.168.2.65:35990      192.168.1.40:1884       ESTABLISHED
tcp        1      0 192.168.2.65:554        192.168.1.40:49126      CLOSE_WAIT
tcp       92      0 192.168.2.65:554        192.168.1.40:49224      CLOSE_WAIT
tcp       92      0 192.168.2.65:554        192.168.1.40:49522      CLOSE_WAIT
netstat: /proc/net/tcp6: No such file or directory
udp        0      0 0.0.0.0:56841           0.0.0.0:*
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node Path
unix  2      [ ACC ]     STREAM     LISTENING         93 /dev/socket/property_service
unix  2      [ ]         DGRAM                       189 /tmp/logsock
unix  2      [ ]         DGRAM                     24518 /var/run/wpa_supplicant-global
unix  2      [ ]         DGRAM                     24534 /var/run/wpa_supplicant/wlan0
unix  3      [ ]         STREAM     CONNECTED         95
unix  3      [ ]         STREAM     CONNECTED      10038
unix  2      [ ]         DGRAM                       722
unix  3      [ ]         STREAM     CONNECTED      10037
unix  2      [ ]         DGRAM                       204
unix  3      [ ]         STREAM     CONNECTED         96

`

roleoroleo, 2020-09-03:
The watchdog wd_rtsp.sh should solve your issue: if rRTSPServer is blocked, it cannot send the stream to the connected clients.
When it's blocked, the CPU goes to 0.0 but there are connected sockets.

In your first ps h264grabber is missing and this is not ok.
I will check wd_rtsp.sh and try to improve it.

n3roGit, 2020-09-03:
Big Thanks!

roleoroleo, 2020-09-13:
Do you want to try this commit manually?
https://github.com/roleoroleo/yi-hack-MStar/commit/f8937c8d76b3411f4619998ba623328492c55755

n3roGit, 2020-09-13:
What's to do to manually change it?

roleoroleo, 2020-09-13:
Edit /home/yi-hack/script/wd_rtsp.sh
and remove the if at line 51:
`    if [ $SOCKET -eq 0 ]; then`
and the corresponding `fi` at line 62.

n3roGit, 2020-09-14:
I changed the file. I give you feedback on some days

n3roGit, 2020-09-17:
So the adjustment has brought an improvement. When the image is gone, it works again after a few minutes.

n3roGit, 2020-09-21:
Still working like charm

n3roGit, 2020-10-02:
Still working. I think you could add this to the actual build :)

roleoroleo, 2020-10-03:
Ok, thanks.

closed 2020-11-21 by roleoroleo