roleoroleo/yi-hack-MStar #216: ONVIF h201c Synology

dmiart, 2020-07-21:
Hi,

I installed the latest (at the time of writing) firmware and the ONVIF doesn't seem to work at all. Tried both Synology and ONVIF device manager v2.2.250 on Windows. Both Server and WS-discovery are enabled in the settings. Am I missing something?

roleoroleo, 2020-07-21:
Sinology has a know issue.
Odm should work correctly.

dmiart, 2020-07-21:
ONVIF device manager works now. Was a firewall issue, but Synology does not still(except for RTSP). Couldn't we just spoof one of the supported devices with similar specs? What other possible ways are there to control the motors on the camera other than ONVIF or through app and website?

roleoroleo, 2020-07-22:
> ONVIF device manager works now. Was a firewall issue, but Synology does not still(except for RTSP). Couldn't we just spoof one of the supported devices with similar specs?

We could take a look at network traffic if someone could capture it with wireshark.
But it is a difficult way.

> What other possible ways are there to control the motors on the camera other than ONVIF or through app and website?

See the wiki about web services.

dmiart, 2020-07-22:
> We could take a look at network traffic if someone could capture it with wireshark.
> But it is a difficult way.
>
>
> See the wiki about web services.

I can do that. Luckily I own  ONVIF camera that works with Synology and know how to use Wireshark. Are you interested in traffic between ODM and the cameras or between Synology and cameras? The latter would be more difficult to capture as I would need to do it in between the two. 

Also, I have attempted to spoof my other camera. It shows up as the same one(except for profiles) and Synology still doesn't like it. I wasn't able to spoof Location tag though. 

[image]

Thanks for the wiki. It's exactly what I am looking for.(control over url), but fixing Onvif would be nice too since lots of people will be using their cameras from Synology directly.  I'll post sniffs later today probably.

closed 2020-11-21 by roleoroleo