roleoroleo/yi-hack-MStar #219: Update 0.3.3 > 0.3.5 failed?

hagenuck1, 2020-07-29:
Hi, 
I had installed 0.3.3 and tried to update yesterday using the web interface. Textfeld showed something similar to „waiting for the camera to come back online“ or so in the end. IP was pingable. But the orange light was on all time and either :8080 webintergace, nor FTP was available. 

Tried downgrade to 0.3.3/ 0.3.4/ 0.3.5 using the SD card, but none of them were working, only ping was possible. 

Installed and tried the YI Camera app now and I can see the image, do anybody of you has any idea how to get mstar installed again? Using an YI 1080, so y203 image.

roleoroleo, 2020-07-30:
Reset the cam and try hack again.

hagenuck1, 2020-07-30:
Is there any other reset then inserting the sd card with another image version of mstar hack? That's what I've tried at least 5 times.

roleoroleo, 2020-07-30:
The upgrade procedure overwrites the flash memory partition.
All files are overwritten

If you want to reset the config too, format the sd before the upgrade.

hagenuck1, 2020-07-30:
Formatting the SD card and flashing 0.3.5 again did the trick, thanks.

closed 2020-07-30 by hagenuck1