roleoroleo/yi-hack-MStar #222: ONVIF preset

amaia81, 2020-08-16:
Hi. Is there a way to set preset positions for the cam without the yi app? Also, How can I extract the PTZ preset profile token to integrate it on an automation in HA? Thanks!

roleoroleo, 2020-08-17:
At the moment it's not possible.

amaia81, 2020-08-18:
Hi roleoroleo, thanks for the quick answer!. 

I would like to be able to send a command from HA to the camera to move to a determined preset position I made at the Yi home APP. In order to do so I've been attempting it with the ONVIF.PTZ preset service but it requires the PTZ preset profile token as per the integration instruction https://www.home-assistant.io/integrations/onvif/ and I'm not beeing able to know how to extract that token.

Thanks!!

roleoroleo, 2020-08-18:
Try the number of the profile.

amaia81, 2020-08-27:
Yes! That solved it! Thanks

closed 2020-08-27 by amaia81