roleoroleo/yi-hack-MStar #223: Other camera support

cjgarciao, 2020-08-18:
Hello,
I have a camera with the MStar 313E chipset. U-boot looks for the files 'sys_h305r' and 'home_h305r'. If I use the 'h201c' files and simply rename them, is it possible that it works or almost certainly not?
Thank you,
Carlos.

roleoroleo, 2020-08-18:
Probably you will brick your cam.

cjgarciao, 2020-08-18:
But I'll be able to restore the copy of the original firmware, right?
Then, should I compile the new firmware, adding the 'h305r' camera? I guess we'll have to change more things.

Thank you,
Carlos.

roleoroleo, 2020-08-18:
If u-boot looks for sys and home partitions, you will be able to restore your cam.
You have to extract a backup copy.

If the fw is very similar, you can compile a new version.
But it must be "very similar".
There are a lot of cams based on the same platform where the porting is complicated:
- different size of flash
- different type of partition (squashfs, jffs2, etc...)
- different framebuffer file

You should start by dumping the flash and extracting a bootlog.

cjgarciao, 2020-08-18:
I think the firmware of my camera is quite similar. The size of flash is 16 Mbytes. In the next link are the bootlog and home and sys partitions (they are jffs2).
https://drive.google.com/drive/folders/16D7-u7PMXaJ4su0IgCVXoXzh6UhJia4s?usp=sharing

Thanks.
Carlos.

roleoroleo, 2020-08-18:
Ok, I downloaded them.
I will compare with the other firmwares.

roleoroleo, 2020-08-23:
Do you want to try this hack?
https://drive.google.com/file/d/1BrVfDqoGYDVmvzQBJ8ASALS4Stqkqcer/view?usp=sharing

cjgarciao, 2020-08-25:
Hello,
It seems to work properly. Then later I will test it well. The camera registers in the cloud and works (and can be deactivated). The web interface works correctly, it seems that the PTZ works too.
The serial console has no shell.
Did you have to make many changes?.
Thank you very much.

roleoroleo, 2020-08-25:
At the moment I only added the cam model into the list of supported cams.
And I removed a scripts that rename japanese audio files because in your version they don't exist.

roleoroleo, 2020-08-25:
Please, test it and give me a feedback.
Particularly the rtsp stream.

What the serial prefix? bfus, 9fus, ... ?

cjgarciao, 2020-08-25:
RTSP in high res works (with a 2-4 seconds of delay). In low res doesn't work. The snapshot works in low res, but not in high res.
The camera is a ieGeek IE80, and the serial (in the web interface) starts by TNPXGAB and finish by BMYVS.

meriororen, 2020-11-05:
Hi @roleoroleo , sorry @cjgarciao I hijack your thread. I have probably similar camera with you, u-boot looking for sys_h305r and home_h305r. I would like to try the hack too, but to be safe I want to backup the stock firmware, can anyone point me where I should read how to dump the firmware on this device? U-boot boot can't be interrupted but I have a linux shell prompt.

roleoroleo, 2020-11-05:
If you have a linux shell, simply dump the mtd partitions with dd.

```
cat /proc/mtd > /tmp/sd/backup/mtd.txt

dd if=/dev/mtdblock0 of=/tmp/sd/backup/mtdblock0.bin
dd if=/dev/mtdblock1 of=/tmp/sd/backup/mtdblock1.bin
dd if=/dev/mtdblock2 of=/tmp/sd/backup/mtdblock2.bin
...
```

closed 2024-05-10 by github-actions[bot]