roleoroleo/yi-hack-MStar #226: Yi app unable to connect despite "disable cloud" being off

kylehase, 2020-08-19:
They Yi app fails to connect most of the time despite the "disable cloud" feature being off. After opening the app and clicking on the camera, the connection will load to 5% then 25% a few times before giving up.

Once it breaks it seems to stay broken even after reboots. What usually fixes the problem temporarily is re-flashing yi-hack-mstar by going down or up a version.

Yi 1080p Dome BFUS
yi-hack-mstar 0.3.5

mar-lor, 2020-08-26:
I had the same issue. In my instance for some reason I didn't have any space left on "/" and the system couldn't create resolv.conf.
I noticed this because "ping" was returning a "bad address" error.
I solved copying /etc over to my sd drive and binding the new directory to /etc at the boot.
Now I can successfully control the camera from the YI cloud app

roleoroleo, 2020-08-26:
Maybe a core dump file created by a crashed program.
Check if there are files with name core.

kylehase, 2020-08-27:
Interesting observation about the full root partition. After checking, I see that my / is full as well. Also, any changes to the hostname or timezone on the configuration page do not persist a reboot.

The output of ps is below and I don't see any core files, which I believe should be in the format 'core.<pid>', but let me know if I should search some other way.

`/ # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.9M      1.9M      4.0K 100% /
tmpfs                    29.7M    128.0K     29.6M   0% /dev
/dev/block/mtdblock3     12.2M     10.7M      1.5M  88% /home
tmpfs                    32.0M     72.0K     31.9M   0% /tmp
/dev/block/mmcblk0p1     29.2G    512.0K     29.2G   0% /tmp/sd

/ # du -hs ./*
597.0K	./bin
512	./data
512	./default.prop
128.0K	./dev
8.5K	./etc
20.0K	./ext
18.1M	./home
512	./init
9.0K	./init.rc
1.8M	./lib
512	./minit
0	./mnt
0	./proc
9.5K	./sbin
0	./sys
584.0K	./tmp
4.0K	./ueventd.rc
31.0K	./usr
512	./var
1.0K	./version.txt

/ # ps
PID   USER     TIME  COMMAND
    1 root      0:02 /init
    2 root      0:00 [kthreadd]
    3 root      0:00 [ksoftirqd/0]
    5 root      0:00 [kworker/0:0H]
    6 root      0:00 [kworker/u2:0]
    7 root      0:01 [rcu_preempt]
    8 root      0:00 [rcu_sched]
    9 root      0:00 [rcu_bh]
   10 root      0:00 [watchdog/0]
   11 root      0:00 [khelper]
   12 root      0:00 [writeback]
   13 root      0:00 [crypto]
   14 root      0:00 [bioset]
   15 root      0:00 [kblockd]
   16 root      0:00 [kworker/0:1]
   17 root      0:00 [kswapd0]
   18 root      0:00 [fsnotify_mark]
   31 root      0:00 [SCLDAZA_THREAD]
   32 root      0:00 [VIPDazaTask]
   36 root      0:00 [deferwq]
   39 root      0:00 /bin/ueventd
   40 root      0:01 [jffs2_gcd_mtd3]
   41 root      0:00 [jffs2_gcd_mtd2]
   78 root      0:00 [mmcqd/0]
   89 root      0:00 [cryptodev_queue]
  100 root      0:00 [spi0]
  122 root      0:00 [kworker/0:2]
  132 root      0:00 [cfg80211]
  137 root      0:00 ./log_server
  138 root      0:01 ./dispatch
  562 root      3:09 ./rmm
  564 root      0:00 [RTW_CMD_THREAD]
  585 root      0:00 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
  594 root      0:00 [kworker/u2:3]
  612 root      0:00 ./mp4record
  613 root      0:00 ./cloud
  614 root      0:01 ./p2p_tnp
  615 root      0:00 ./oss
  616 root      0:00 ./watch_process
  625 root      0:00 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
  639 root      0:00 telnetd
  670 root      0:00 pure-ftpd (SERVER)
  687 root      0:00 udhcpc -i wlan0 -b -s /home/app/script/default.script
  698 root      0:00 dropbear -R
  703 root      0:00 ipc_multiplexer
  729 root      0:00 dropbear -R
  731 root      0:00 mqttv4
  749 root      0:00 -sh
  884 root      0:00 ntpd -p pool.ntp.org
  931 root      0:03 rRTSPServer_audio
  935 root      0:00 {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
  973 root      0:00 onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.3.5 --hardware_id BFUS --serial_num BFUSY1EF7ERARM200311 --ifs wlan0 --port 80 --scope onvif://www.onvif.org/P
  984 root      0:00 wsdd --if_name wlan0 --type tdn:NetworkVideoTransmitter --xaddr http://%s --scope onvif://www.onvif.org/name/Unknown onvif://www.onvif.org/Profile/Streaming
  990 root      0:00 /usr/sbin/crond -c /var/spool/cron/crontabs/
 1013 root      0:00 ./arp_test
 1050 root      0:00 sleep 10
 1057 root      0:00 ps

/ # find / -iname core*
/sys/devices/system/cpu/cpu0/topology/core_id
/sys/devices/system/cpu/cpu0/topology/core_siblings
/sys/devices/system/cpu/cpu0/topology/core_siblings_list
/sys/module/mmc_core/coresize
/sys/module/ccm/coresize
/sys/module/fat/coresize
/sys/module/ssp_ms41909/coresize
/sys/module/mmc_block/coresize
/sys/module/cfg80211/coresize
/sys/module/vfat/coresize
/sys/module/usb_common/coresize
/sys/module/mdrv_mmfe/coresize
/sys/module/mdrv_mvhe/coresize
/sys/module/mstar_jpe/coresize
/sys/module/cpld_periph/coresize
/sys/module/pid_list/coresize
/sys/module/ehci_hcd/coresize
/sys/module/spi_infinity/coresize
/sys/module/usbcore/coresize
/sys/module/8188fu/coresize
/sys/module/mdrv_crypto/coresize
/sys/module/kdrv_sdmmc/coresize
/sys/module/mdrv_cryptodev/coresize
/proc/sys/net/core
find: /proc/1959: No such file or directory
`

roleoroleo, 2020-08-27:
This is my cam.
```
/ # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.9M      1.8M     28.0K  99% /
tmpfs                    29.7M      1.8M     27.9M   6% /dev
/dev/block/mtdblock3     12.2M     10.7M      1.5M  87% /home
tmpfs                    32.0M    176.0K     31.8M   1% /tmp
/dev/block/mmcblk0p1      1.8G      1.1G    748.1M  60% /tmp/sd
```

Probably I have to reduce busybox footprint.

Rushmed, 2020-08-28:
Thats mine, having the same issue:
`/ # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.9M      1.9M      4.0K 100% /
tmpfs                    29.7M    128.0K     29.6M   0% /dev
/dev/block/mtdblock3     12.2M     10.7M      1.5M  88% /home
tmpfs                    32.0M    156.0K     31.8M   0% /tmp
/dev/block/mmcblk0p1     59.6G     36.3G     23.3G  61% /tmp/sd
`

roleoroleo, 2020-08-28:
I think I can remove some applet from busybox to increase the free space.
Next release...

Mdleal, 2020-09-09:
Same issue. Root is full. DNS is not working. Resolv.conf is not coping to /etc/

pejotigrek, 2020-09-11:
> Same issue. Root is full. DNS is not working. Resolv.conf is not coping to /etc/
> 
> Till this is resolved here is a temporary fix.
> 
> cp -r /etc/ /tmp/sd/
> mount --bind /tmp/sd/etc /etc

that didn't helped in my scenario. I used the same commands and still no yi-app connection.

```
/home/yi-hack # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.9M      1.9M      4.0K 100% /
tmpfs                    29.7M    128.0K     29.6M   0% /dev
/dev/block/mtdblock3     12.2M     10.7M      1.5M  88% /home
tmpfs                    32.0M     76.0K     31.9M   0% /tmp
/dev/block/mmcblk0p1    963.0M    128.0K    962.9M   0% /tmp/sd
/dev/block/mmcblk0p1    963.0M    128.0K    962.9M   0% /etc
```

roleoroleo, 2020-09-11:
I removed some apps.
https://github.com/roleoroleo/yi-hack-MStar/commit/deda51ba4f6e3165a9e3e35330e36691fcaafb10

Mdleal, 2020-09-11:
> that didn't helped in my scenario. I used the same commands and still no yi-app connection.
>
> ```
> /home/yi-hack # df -h
> Filesystem                Size      Used Available Use% Mounted on [...]

cp /tmp/resolv.conf /etc/

This does not survive a reboot.

roleoroleo, 2020-09-12:
What your model?
I will send you a beta version.

pejotigrek, 2020-09-13:
if you're asking me - yi dome 1080p, model suffix h201c,
base version 4.6.0.0A_201908271549, serial number starts with BFUS

steef84, 2020-09-15:
Same problem as mentioned for my y203c cams. May I try your beta version?

roleoroleo, 2020-09-15:
I will publish a new release.

pgallent, 2020-09-15:
Thanks, same problem here.

roleoroleo, 2020-09-16:
> Same problem as mentioned for my y203c cams. May I try your beta version?

You have the same camera as me.
You should have a lot of free space.
There are no processes writing to rootfs, it should never change.
Check if a core file filled the space.
```find / -name core*
```

bepece1, 2020-09-16:
No space left on my device (h201c) (yihack 0.3.5)
What can I do?

I've tried with this command but it no work:
```
cp -r /etc/ /tmp/sd/
mount --bind /tmp/sd/etc /etc
```

Thanks

roleoroleo, 2020-09-16:
Waiting for a new release you could reflash the current release.
Downgrade to the previous and upgrade again.

steef84, 2020-09-16:
```
sys/devices/system/cpu/cpu0/topology/core_id
/sys/devices/system/cpu/cpu0/topology/core_siblings
/sys/devices/system/cpu/cpu0/topology/core_siblings_list
/sys/module/mmc_core/coresize
/sys/module/ccm/coresize
/sys/module/fat/coresize
/sys/module/mmc_block/coresize
/sys/module/cfg80211/coresize
/sys/module/vfat/coresize
/sys/module/usb_common/coresize
/sys/module/mdrv_mmfe/coresize
/sys/module/mdrv_mvhe/coresize
/sys/module/mstar_jpe/coresize
/sys/module/cpld_periph/coresize
/sys/module/pid_list/coresize
/sys/module/ehci_hcd/coresize
/sys/module/spi_infinity/coresize
/sys/module/usbcore/coresize
/sys/module/8188fu/coresize
/sys/module/mdrv_crypto/coresize
/sys/module/kdrv_sdmmc/coresize
/sys/module/mdrv_cryptodev/coresize
/proc/sys/net/core
find: /proc/3257: No such file or directory
```

roleoroleo, 2020-09-16:
28.0K available is normal.
The root partition is not used to write.

steef84, 2020-09-17:
Oke but still got 5 (!) y203c cams with the problem unable to connect with Yi app.
For me it occurs after being away for 5 weeks, and all cams were off. For now I can acces local cam webinterface, and rtsp stream, ssh and all,but cannot acces via Yi App.
Same for me also cant ping google as mentioned by @mar-lor .Even your update button fails. My cams running 0.3.4. 

```
/home/yi-hack # ping google.com
ping: bad address 'google.com'
```

[image]

roleoroleo, 2020-09-17:
If you restart the cams, do they reconnect?
If yes, your problem is not related to the free space on root partition.

roleoroleo, 2020-09-17:
0.3.6 released with reduced busybox.

steef84, 2020-09-18:
pinging 8.8.8.8 indeed worked on the cams. But they couldnt find updates in webinterface (error) and they couldnt connect to YI.

Now I've updated 4 manually to 0.3.6 and the instantly opens again in Yi App. Can I provide more info for you?

roleoroleo, 2020-09-18:
If the ping to 8.8.8.8 works the problem is /etc/resolv.conf
A correct dns entry is missing.
Try the cams with the new fw and give me a feedback in a few days.

steef84, 2020-09-27:
To report back, cams are still online and connected with Yi Cloud. Great!!

kylehase, 2020-09-27:
Same here. Installed 0.3.6 on the day it was released (10 days ago) and the problem hasn't occurred since.

closed 2020-11-21 by roleoroleo

denven, 2020-12-02:
V0.3.8
`/ # uptime
 23:57:34 up 1 day, 20:45,  load average: 4.84, 4.85, 4.96
/ # df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.9M      1.9M      8.0K 100% /
tmpfs                    29.7M      1.8M     27.9M   6% /dev
/dev/block/mtdblock3     12.2M     10.7M      1.5M  88% /home
tmpfs                    32.0M    184.0K     31.8M   1% /tmp
/dev/block/mmcblk0p1     29.7G     17.8G     11.9G  60% /tmp/sd
/ #`

This is a fatal as last time I lost control of my camera(cannot pair with the app), worrying about aother time.

roleoroleo, 2020-12-02:
I will try to delete all writes on the filesystem.
To solve the problem, downgrade to the previous version and upgrade again.

pinguinpfleger, 2021-01-21:
same here FW 0.3.6