roleoroleo/yi-hack-MStar #228: How the h264grabber work

mendellev82, 2020-08-20:
Hi @roleoroleo can you explain how the h264grabber work?

From your info there is a buffer file at
/dev/fshare_frame_buf

But according to the source code in h264grabber.c
using the files in
/proc/mstar/OMX/VMFE1/ENCODER_INFO/OBUF_pBuffer
and
/proc/mstar/OMX/VMFE1/ENCODER_INFO/OBUF_nAllocLen

Is the above two consider a file? not a process?
Under /proc is not a process?

Some more there is a devie /dev/mem and what is the purpose?

So in conclusion my questions are
1. What are the general steps (general flows) to generate /tmp/h264_high_fifo file from the above mention files?
    For example list the steps and add some explanation to each steps. 
2. What are the relationship between them such as fshare_frame_buf, OBUF_pBuffer and /dev/mem?
3. Is that the audio stream is included in /tmp/h264_high_fifo?

Out of scope questions:
1. What IDE or editor is usually use to write the C/C++ for compile in Linux?
2. After we write the C/C++ source code, how we generate the config / make file?
   What tools we need to use?

I am a beginner that learning the Linux, I hope you can my doubts.
I know I can simply use the release but that is not my purpose.
I hope to learn something from this project too.
Sorry for long questions.
Thank you and look for your reply.

roleoroleo, 2020-08-20:
/dev/fshare_frame_buf is a file used by Yi to store a circular buffer of the video stream.
But in this platform I don't use it because there is a easier method.
fshare_frame_buf is used instead in Allwinner platform.

The /proc filesystem is a virtual filesystem where the kernel exposes informations and where you can interact with the kernel from userspace.
In our case, whoever developed the kernel driver that takes care of the h264 encoding, created some files that contain important informations.
/proc/mstar/OMX/VMFE1/ENCODER_INFO/OBUF_pBuffer contains the memory address of the buffer where the driver saves the encoded frame and
/proc/mstar/OMX/VMFE1/ENCODER_INFO/OBUF_nAllocLen contains the len of this buffer.

Using the file /dev/mem you can have access to all the memory and therefore, knowing the address, you can have access to the frames.

> So in conclusion my questions are
> 
>     1. What are the general steps (general flows) to generate /tmp/h264_high_fifo file from the above mention files?
>        For example list the steps and add some explanation to each steps.

Continuously read the buffer where the frames are stored and write them on the fifo.

>     2. What are the relationship between them such as fshare_frame_buf, OBUF_pBuffer and /dev/mem?

fshare_frame_buf is not used in this platform. But it's a file where rmm process save the video stream, both high and low res.

>     3. Is that the audio stream is included in /tmp/h264_high_fifo?

No, the audio stream is not included.
The audio stream is muxed directly from rtsp daemon.

> Out of scope questions:
> 
>     1. What IDE or editor is usually use to write the C/C++ for compile in Linux?

If the project is simple (2 or 3 C files) I use only a text editor like Midnight Commander and I write Makefile manually.
If the project is bigger I use Eclipse.

>     2. After we write the C/C++ source code, how we generate the config / make file?
>        What tools we need to use?

Manually or Eclipse.

>     3. For example we want to cross compile the binary to run on Yi camera on x86 Linux PC
>        is that all the require toolchain are incuded in SDK or we can simply download a
>        standard toolchain in x86 Linux PC directly use it?

The toolchain is a very complicated thing. If you want to run a binary in the cam without compile statically (saving space) and without add other libc you must use a toolchain based on the same libraries.
Same libc, same gcc (or not too different), etc...
There are 2 possibilities:
1. You find the toolchain googling on the net (typically the toolchain is made by whoever produces the microprocessor).
2. You create your own toolchain using a tool like crosstool-ng (but you have to collect all the information you need).

>     4. Got such a thing that standard toolchain for each platform
>        such as for arm? If yes, we can use that standard toolchain to cross compile for any Linux run on arm?
>        If no is that in order to compile for each of the embedded Linux we need to download the corresponding SDK which include
>        the toolchains? Mean even though for different Linux system run on ARM, they can't simply share the toolchain?

Toolchains are very different and typically not compatible with each other.
If you use a toolchain other than the original one, you have to compile the programs statically.
This means that the programs become very large and do not normally fit in the remaining flash space.

I hope I have answered the questions clearly.
If you have other questions, please ask.

mendellev82, 2020-08-20:
@roleoroleo Thank you for your reply. It really provide  a great help for me to understand the project. According to the similar project such as YihackV4 I don't see something similar to /proc/mstar/OMX/VMFE1/ENCODER_INFO/OBUF_pBuffer is that mean for that version of cam, it not using that method that store useful info in those files right? Last question, are you know those related files such as /proc/mstar/OMX/VMFE1/ENCODER_INFO/OBUF_pBuffer through reverse engineering or else? There must be some method you can find out those related files. Thanks again.

roleoroleo, 2020-08-20:
This method using /proc files works only for this cam because the driver exposes these informations.
Yi-hack-v4 is made for a platform that doesn't have /proc/mstar/* info.

The rtsp daemon streams 3 channels:
1. ch0 - high res video + audio (if enabled)
2. ch1 - low res video + audio (if enabled)
3. ch2 - only audio (if enabled)

I found the /proc files via reverse engineering.
Studyng the file system, using tools like lsof and strace and a lot of time...

mendellev82, 2020-08-20:
There is a long way for me to study.
Anyway thank you so much.
Any method to contact you such as email?
Some of the projects become inactive and is hard to contact the owner.
Or github support private message?

roleoroleo, 2020-08-20:
You can see my email address clicking on paypal donate button.

mendellev82, 2020-08-20:
Ok thanks