roleoroleo/yi-hack-MStar #230: FTP directory

Arkezis, 2020-08-24:
Hi,
as a french user of your awesome hack, I had to modify the ftppush code to allow upload on the router of my internet provider.
It does allow upload only on "Freebox" and I did put them in "camera" folder. I only needed to update a line of the code (and another to upload only the file with all folders concatenated in the name) but I think other people could have the same issue.
Is it possible to add a textfield for the next version to set the "destination path" ? 

Thanks :)

roleoroleo, 2020-08-24:
Yes, probably to the next version.
See this issue:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/101

The platform is different but the source is the same.

Arkezis, 2020-08-24:
Awesome thanks ! 
What's the difference between the 2 repo ? oÔ

roleoroleo, 2020-08-24:
Different hardware.

janekjelonek, 2020-09-18:
Hello, 
I have a feeling that files pushed via FTP gets overwritten in destination directory because of the non-unique names ex.: 29M00S60.mp4. Full year/month/day/hour/minute path would fix it.  (0.3.6 version)

take care, Janek

roleoroleo, 2020-09-19:
You are right.

karol93, 2020-09-20:
i have the same problem with the name of files when upload to ftp is enabled. more details in files name will be an amazing feature ;)

julk77, 2020-09-22:
Hello,
I have made some little changes on the ftppush.sh script, so that the folders are created on the ftp server, the same way than they would have been created locally.
Please test it and let me know:
https://github.com/julk77/yi-hack-MStar/blob/patch-1/src/static/static/home/yi-hack/script/ftppush.sh

roleoroleo, 2020-09-22:
Thank you.
I will test it asap.

roleoroleo, 2020-09-22:
Tested successfully.
I will add it with a new option: "Create directory tree".

roleoroleo, 2020-09-24:
https://github.com/roleoroleo/yi-hack-MStar/commit/e941ad06e77cc3a57da772177ae7f2cb25927785

braxanmp, 2020-10-07:
Hi! I would like to upload these files in the same folder but with the YYYYMMDD include in the name. I have tried to modify the ftppush.sh but i never used shell... I think I should edit this line:
`	if ( ! ftpput -u "${FTP_USERNAME}" -p "${FTP_PASSWORD}" "${FTP_HOST}" "/${FTP_DIR}${FTP_DIR_HOUR}$(lbasename "{UTF_FULLFN}")" "${FTP_DIR_HOUR}$" ); then
		echo "[ERROR] uploadToFtp: ftpput FAILED."`

Could you help me to modify it correctly? Thank you in advance!

julk77, 2020-10-07:
> Hi! I would like to upload these files in the same folder but with the YYYYMMDD included in the file name. I have tried to modify the ftppush.sh but i never used shell... I think I should edit this line:
> ` if ( ! ftpput -u "${FTP_USERNAME}" -p "${FTP_PASSWORD}" "${FTP_HOST}" "/${FTP_DIR}${FTP_DIR_HOUR}$(lbasename "{UTF_FULLFN}")" "${FTP_DIR_HOUR}$" ); then echo "[ERROR] uploadToFtp: ftpput FAILED."`
> 
> Could you help me to modify it correctly? Thank you in advance!

Hi,

Lines 146 and 147 of https://github.com/julk77/yi-hack-MStar/blob/patch-1/src/static/static/home/yi-hack/script/ftppush.sh

Replace 
echo -e "USER ${FTP_USERNAME}\r\nPASS ${FTP_PASSWORD}\r\nmkd ${FTP_DIR}/${FTP_DIR_HOUR}\r\nquit\r\n" | nc -w 5 ${FTP_HOST} 21 | grep "${FTP_DIR_HOUR}"
FTP_DIR_HOUR="${FTP_DIR_HOUR}/"
With
#echo -e "USER ${FTP_USERNAME}\r\nPASS ${FTP_PASSWORD}\r\nmkd ${FTP_DIR}/${FTP_DIR_HOUR}\r\nquit\r\n" | nc -w 5 ${FTP_HOST} 21 | grep "${FTP_DIR_HOUR}"
FTP_DIR_HOUR="${FTP_DIR_HOUR}_"

This should do what you want...

closed 2020-11-21 by roleoroleo