roleoroleo/yi-hack-MStar #231: YI Home 1080 BFUS

arekdudys, 2020-08-31:
Hi! YI Home 1080 (BFUSY31) is skipping the firmware (y203c_0.3.5) update process at all. It's booting normally Yellow freezed LED for few second then it turns blue. The only thing that comes to my mind - After installing Yi Home app, i received firmware update and I installed it. It is possible that's the case?

Few minutes earlier I did it successfully for the Mi Dome 1080 (So I am assuming its not related to the sd card)

Any tip highly appreciated :)
// thanks for the amazing job. I really happy how it is working with Dome!

roleoroleo, 2020-08-31:
Probably it's a different model.
Are you sure it's a y203?

arekdudys, 2020-08-31:
Right, i saw the `BFUS` so i thought thats the one.  It seems I have a newer generation one. its called Yi Home Camera 1080P AI+ , and it has "AI+" logo at rear side. Also the firmware is 8.2.0 ...  Sorry for making mess

roleoroleo, 2020-08-31:
Use Allwinner hack.

closed 2020-09-13 by roleoroleo