roleoroleo/yi-hack-MStar #233: Mqtt messages

bonuzzz, 2020-09-07:
Is it possible to add to mqtt birth and last will messages?

roleoroleo, 2020-09-07:
Could you explain me better?

bonuzzz, 2020-09-07:
when device starts working, it sends to mqtt birth message like online for example. when disconnected, rebooted or something else, last will message would be sent, "offfline" for example. it could be simplier monitor camera status.
for implementation probably it would help https://github.com/eschava/broadlink-mqtt/commit/0c34ae43837510ef88c0eb7dad229c568f324db4 but it uses python library.

bonuzzz, 2020-09-07:
birth_message: The message to send when a connection to the broker is established.
will_message: The message to send when the MQTT connection is dropped.

roleoroleo, 2020-09-09:
I think I can add it.
But the will message probably will not work.
mqttv4 doesn't have signals management, when you switch off the cam (hard or soft) mqttv4 terminates without a clean disconnection.

foxel, 2020-09-09:
@roleoroleo you don't need to bother about catching the offline status by somehow. Last will message is provided by the client while it's still online and the broker keeps it safe until the connection to the client is closed for any reason. Then the last will message is delivered by the broker.

roleoroleo, 2020-09-09:
Added: https://github.com/roleoroleo/mqttv4/commit/446d9f10791a5b5248160a302fb52251dba9c0c0

foxel, 2020-09-09:
So... I think the main concept is missing here.

MQTT last will message is used to track the device state. It's usually used in a combination with birth message.
Most of the cases both messages are sent to the same topic and state the device status:
* birth message reads 'online' or '1', etc.
* last will reads 'offline' etc.

For some cases or flexibility sometimes the topics are flexible and can be separated.
See HA configuration for example: https://www.home-assistant.io/docs/mqtt/birth_will/
I recommend having your defaults covering this UC:
* TOPIC_BIRTH == TOPIC_WILL
* BIRTH_MSG = 'online'
* WILL_MSG = 'offline'

Or maybe you can consider not having this so flexible and allow the user to select only one topic for both messages. Same way as it's done with motion detection topic.

Looking thru the changes I see that the implementation of the last will message is incorrect.
The main feature of MQTT's last will is to be sent on any connection drop. And to achieve this the message is sent at the connect step: [MQTT protocol. chapter 3.1 "CONNECT"](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Ref363033523) - read 3.1.2.5 - 3.1.2.7. The broker will deliver the message on the client disconnect. That's why it's named "last will" :)

bonuzzz, 2020-09-09:
Agreed with  @foxel   They're should be in one topic, for example State, and two messages: online and offline

roleoroleo, 2020-09-10:
Thank you for the explanation.

What happens when the disconnection is gracefully, is it necessary to send an explicit will message before disconnect or does the broker send it?

bonuzzz, 2020-09-10:
Broker sends it

foxel, 2020-09-10:
> Thank you for the explanation.
> 
> What happens when the disconnection is gracefully, is it necessary to send an explicit will message before disconnect or does the broker send it?
> 

According to the document I mentioned above. Of client sends DISCONNECT server must discard last will message. So if you do clean disconnect you should send offline status message before

foxel, 2020-09-13:
@roleoroleo I was trying to compile the code for my desktop with no success... can you help me with that or send x64 compiled binary to test?

roleoroleo, 2020-09-13:
Remove config.mak and run make.
You will see an error:
"make[1]: mqttv4: Command not found"
but the binary is built.

Check the dependencies: libmosquitto-dev for example.

bonuzzz, 2020-09-13:
@foxel 
https://drive.google.com/file/d/1bwtM1T2aLzjl3fJOd1IHSybDOQSJGJUe/view?usp=sharing
this is compiled binary

foxel, 2020-09-13:
@roleoroleo thanks for the guide. I managed to find the mistake: `mqtt_conf->topic_birth_will` contains suffix only.
I made the following and everything worked as expected:
```c
    char topic_birth_will[128];
    sprintf(topic_birth_will, "%s/%s", mqtt_conf->mqtt_prefix, mqtt_conf->topic_birth_will);
    ret=mosquitto_will_set(mosq, topic_birth_will, strlen(mqtt_conf->will_msg), mqtt_conf->will_msg, mqtt_conf->qos, mqtt_conf->retain_birth_will == 1);
```

P.S. I had segfault until filled all the config variables in... Looks like the binary has no defaults and tries to run even if the config is incomplete. Maybe you should add some validation in `handle_config` to make sure that the program can continue and give the user a reasonable message otherwise...

@bonuzzz I appreciate this

roleoroleo, 2020-09-13:
The problem was under my eyes but I couldn't see it.
I will fix this problem and I will add a default for all parameters.

Thanks for your contribution.

roleoroleo, 2020-09-13:
Ok.
Fixed and tested.
https://github.com/roleoroleo/mqttv4/commit/83f7a8f42e31ad6f0a688785b10049b39afc5df3
https://github.com/roleoroleo/mqttv4/commit/5f77700a9e8317712c0b66989bf9224f21063589
I will include it in the next release.

bonuzzz, 2020-09-14:
Probably it's strange, but I see only birth messages with latest code

roleoroleo, 2020-09-14:
I tested it again.
It seems to work.

bonuzzz, 2020-09-14:
@roleoroleo could you upload your binary?
I can't flash whole firmware. It hangs on startup, so now I'm testing with binary replacement

bonuzzz, 2020-09-14:
two issues:
1. I have will message outside any topics on mqtt, on parent folder
2. I have will message only when I manually break mqttv4 process. I don't see anything when reboot camera from web interface

roleoroleo, 2020-09-14:
1. Check the prefix topic, the code is ok.
2. If I reboot the cam from web interface, the will message appears after 20-30 seconds.
Probably the mqtt broker needs time...

bonuzzz, 2020-09-19:
@roleoroleo I think something wrong in current implementation. I found this article. Probably I will help https://medium.com/@emqtt/use-of-mqtt-will-message-b1686aab6e95
And I think it's better to remake will message and implement through mosquitto_sub like in example in that article. I tested it and works flawlessly.

roleoroleo, 2020-09-19:
I will check it.

foxel, 2020-09-19:
@roleoroleo I can confirm 0.3.6 is broken. I also checked what version is referenced from the 3.6.0 tag and it's before fix with parent topic: navigate to mqttv4 from here https://github.com/roleoroleo/yi-hack-MStar/tree/0.3.6/src/mqtt

@bonuzzz mosquitto_sub is a desktop client implementation, that's just an example. I see no reason for putting another binary to the cam

roleoroleo, 2020-09-19:
Sorry but these days I have little time and I'm getting things done quickly.
I didn't pull the submodule.
Please try this updated binary:
[mqttv4.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5250430/mqttv4.gz)

foxel, 2020-09-19:
The message appears when camera is turned off but only after keepalive timeout which is 120 seconds by default. That's also true for reboot. Also see #240

foxel, 2020-09-19:
> Sorry but these days I have little time and I'm getting things done quickly.
> I didn't pull the submodule.
> Please try this updated binary:
> [mqttv4.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5250430/mqttv4.gz)

@roleoroleo this binary works well :+1:

bonuzzz, 2020-09-19:
@foxel  camera has mosquitto_pub and mosquitto_sub, so nothing is needed to compile more.
@roleo now will message is in proper topic, but implementation is still wrong. As I understand, now mqtt binary sends will message when process killed or finished. But when network connection drops, nothing happens. Will message should be cached on mqtt broker and it will show when connection to mqtt client dropped by any reason.

foxel, 2020-09-19:
@bonuzzz no. When killed the broker delivers the message. It happens when keepalive ping fails.

bonuzzz, 2020-09-19:
@foxel but when network connection dropped (I turn off wifi on router), nothing happens

foxel, 2020-09-19:
@bonuzzz what's your 'Keepalive' value set in 'Advanced' ?

roleoroleo, 2020-09-19:
> @foxel but when network connection dropped (I turn off wifi on router), nothing happens

Are the subscribers connected to the same wifi?
:)

bonuzzz, 2020-09-19:
@foxel  I missed that parameter.  120 sec. Set to 10 sec and now it works as expected. Thanks
@roleoroleo of course (if I understand your question). I just missed keepalive parameter. So everything is ok now.
If nobody don't mind, I will close this issue.
Thanks to everyone:)

closed 2020-09-19 by bonuzzz