roleoroleo/yi-hack-MStar #235: YI 1080p Home YYS.2016 impossible to hack

josemabcn, 2020-09-08:
Dear all,

I have have puchased the YI 1080p Home from amazon Spain, the camera have the reference YYS.2016 and the firmare is : 8.2.0.0A_202007211802 and is impossible to update the firmware hack. The yellow red is not blinking and 2 oe 3 seconds after the blue is continuously on. It's impossible to hack this cam version?.

Thnaks,

Jose

neolution, 2020-09-08:
Please try this version: https://github.com/roleoroleo/yi-hack-Allwinner/blob/master/README.md

closed 2020-09-13 by roleoroleo

avellma, 2022-08-28:
Hello @neolution ..does the link that you provided supports the YYS.2016 ???
If so, which file should I use ???

Camera | Firmware | File prefix | Remarks
-- | -- | -- | --
Yi 1080p Home 9FUS | 8.2.0* | y20ga | -
Yi 1080p Home BFUS | 8.2.0* | y20ga | -
Yi 1080p Home BFCN | 8.2.0* | y20ga | -
Yi 1080p Home 9FUS | 8.3.0* | y25ga | beta version
Yi 1080p Home IFUS | 8.3.0* | y25ga | beta version
Yi Dome X BFUS | 8.1.0* | y30qa | beta version

It is not clear to me which one is the right firmware for the camera I have
What I have is the YYS.2016, model YI1080P Home, with ID RFUSY620D7 ....

[image]

roleoroleo, 2022-08-28:
Check the fw version and read this page:
https://github.com/roleoroleo/yi-hack-Allwinner-v2

avellma, 2022-08-29:
Hello @roleoroleo I did try the yi-hack-Allwinner-v2, but I did not work.
My camera is:

Yi 1080 home
Model Number: YYS.2016.CN
Firmware version: 11.1.35.3_202207191613
The SD card is a 16gb and formatted as recommended SD Card as FAT32

This is what I loaded to the SD Card
[image]

greagank, 2022-12-16:
y291g worked perfectly for me

josequinta-pixelarus, 2023-04-25:
> y291g worked perfectly for me

That version worked for me too! Thank you!!

mvastum, 2023-06-15:
> > y291g worked perfectly for me
> 
> That version worked for me too! Thank you!!

Could you send me a link to that firmware? I’m only able to find y291ga and y29g, not y291g. I have the exact same camera model (YYS.2016) and would like to enable RTSP. Thanks a lot in advance.

bullyhupe, 2023-09-22:
Hello,
I can't find the firmware.
Could you send me a link to firmware y291g too?

biggest thanks

roleoroleo, 2023-09-26:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/tag/0.2.9

F2Jab, 2024-11-30:
> y291g отлично мне подошел

> > y291g отлично мне подошел
> 
> У меня тоже эта версия сработала! Спасибо!!

> > > y291g отлично мне подошел
> > 
> > 
> > У меня тоже эта версия сработала! Спасибо!!
> 
> Не могли бы вы прислать мне ссылку на эту прошивку? Я могу найти только y291ga и y29g, но не y291g. У меня точно такая же модель камеры (YYS.2016) и я хотел бы включить RTSP. Заранее большое спасибо.

> Здравствуйте, не могу найти прошивку. Не могли бы вы мне тоже прислать ссылку на прошивку y291g?
> 
> огромное спасибо

> https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/tag/0.2.9

Hello, can you tell me where to find this particular firmware y291g? Or did you make a mistake and put y291ga?! There is no y291g in all releases. Or did the developer remove it? Thank you all!

roleoroleo, 2024-12-13:
Please, make a backup and share it with me.
I'll check the version.