roleoroleo/yi-hack-MStar #236: Just to 5FCN is also supported

leioo2000, 2020-09-09:
Just to report my 5FCN Yi home1080 also works well with firmware y23, considering I didn't see it in the "supported cameras" list. Thank you so much for your great work!

It works well including both RTSP (with audio) and snapshots! I didn't find proxy settings in y23 firmware (camera keeps reminding this camera can only be used within China in the first few minutes after rebooting, and it seems it can't remember the wifi connection, so must add it to wifi manually everytime after rebooting too), so far there is no way to add this camera to the Yi App, but for me it's ok, it's already much better than putting it in the corner to be covered in dust.

leioo2000, 2020-09-09:
Correction, my bad, proxy is there I just missed it, setting it is the save as in yi-hack-v4. However so far there is no luck to bundle it to the yi app, gonna try it more times.

leioo2000, 2020-09-09:
device bundled successfully! the cause seems like I disabled cloud. After enabling cloud it works all fine now.

easion, 2020-09-28:
寻在中国大陆买到mstar方案如5FCN/6FCN 的设备名称或者淘宝、京东等平台的购买网址,有知道的请回复下，谢谢.

leioo2000, 2020-10-01:
> 寻在中国大陆买到mstar方案如5FCN/6FCN 的设备名称或者淘宝、京东等平台的购买网址,有知道的请回复下，谢谢.

我是在淘宝小蚁官方旗舰店买到的5FCN，但已经是3年前的事了，不知道现在还是不是仍旧是5FCN的。产品链接如下：
https://m.tb.cn/h.4aglrNN?sm=43ef88
注：建议和卖家沟通确认这个仍旧是5FCN后再下单。

easion, 2020-10-12:
> > 寻在中国大陆买到mstar方案如5FCN/6FCN 的设备名称或者淘宝、京东等平台的购买网址,有知道的请回复下，谢谢.
> 
> 我是在淘宝小蚁官方旗舰店买到的5FCN，但已经是3年前的事了，不知道现在还是不是仍旧是5FCN的。产品链接如下：
> https://m.tb.cn/h.4aglrNN?sm=43ef88
> 注：建议和卖家沟通确认这个仍旧是5FCN后再下单。

感谢！ 已经采购。客服也不清楚型号，他们的电商和仓库是分离的，先试试看，但愿能用！

closed 2020-11-21 by roleoroleo