roleoroleo/yi-hack-MStar #237: yi dome 1080 bdfu

guyyar89, 2020-09-11:
hey,

at 1 of week i need to do downgrade version and after upgrade version beacuse i cant use with the cloud camera,
i need to do this beacuse i get error:
-3030 connection and i dont have problem with network and free space on sd card.

how to fix it?

roleoroleo, 2020-09-13:
What's model and what's base fw?

guyyar89, 2020-09-16:
> What's model and what's base fw?

base version is:
4.6.0.0A_201908271549

model is:
h201c

Firmware Version is:
0.3.5

roleoroleo, 2020-09-16:
Please, explain me better the problem.

guyyar89, 2020-09-16:
> ‫ב-16 בספט׳ 2020, בשעה 23:35, ‏‏roleo ‏<notifications@github.com> כתב/ה:‬
>
>
> Please, explain me better the problem.

When i want to use with application i get error -3030 error conection

roleoroleo, 2020-09-17:
Is "Disable cloud" switched off?

guyyar89, 2020-09-17:
> Is "Disable cloud" switched off?

yes Disable Cloud is off

roleoroleo, 2020-09-17:
Login with ssh and check if you have an internet connection.

closed 2020-11-21 by roleoroleo