roleoroleo/yi-hack-MStar #239: Is any way for making a timelapse?

knut1er, 2020-09-18:
Is it interesting to make a timelapse. I believe it is possible on Yi Cloud.
Many thanks @roleoroleo  for your work!

roleoroleo, 2020-09-18:
I think we need ffmpeg to create mp4 files.
It's too big to embed it in the fw.

knut1er, 2020-09-18:
I should schedule shots & transfers with SSH?

roleoroleo, 2020-09-18:
Yes, you could schedule shots with crond or externally calling the snapshot web service.

knut1er, 2020-09-18:
I did not see the web service feature. That’s great!
Thank you

Dmario-dev, 2020-09-22:
Also looking forward for this feature.
roleoroleo can you provide instruction how to save image to sd card with crond?
When I try to run snapshot.sh from SSH it only saves small white square.

roleoroleo, 2020-09-22:
See the wki.

Dmario-dev, 2020-09-23:
@roleoroleo sorry but there is no description of snapshot.sh file.
When I try:
`/home/yi-hack/www/cgi-bin # ./snapshot.sh high >> ../pages/1.jpeg`
The picture is only small white square. Can you please help with this?

roleoroleo, 2020-09-23:
https://github.com/roleoroleo/yi-hack-MStar/wiki/Web-services-description#snapshot

closed 2020-10-23 by roleoroleo