roleoroleo/yi-hack-MStar #243: Is it possible to change firmware from chinese to global?

maximser, 2020-09-26:
Hi, have a yi 1080p home camera (chinese version, install yi-hack-MStar working great). So, Is it possible to change firmware from chinese to global? Can some one briefly explain why it cant be done(different hardware, id/mac checks or some current firmware/bootloader prevents), or maybe it can be done by directly connecting to memory/eeprom and  flashing it on.

roleoroleo, 2020-09-26:
I don't know exactly but probably it's only the serial number.
You could try to change it.

maximser, 2020-09-26:
I`m asking this because when using chinese firmware (currently lastest for yi 1080p home 2.1.0.0E_201809191630) I have some difficulties: need to find  list of chinese proxy to setup camera, need to change timezone to correct date watermark on video(and it kills the correct work of yi app on phone, i read about that issues here ). So i think global firmware illuminate this unnecessary difficulties.
One more question, on global yi website I can download latest 2.1.0.0E_201809191630home_y20m firmware, is just put it on sd and reboot is enough to flash it?

roleoroleo, 2020-09-26:
If you are using MStar hack with 2.1.0 fw, probably your cam is a y23.
The firmware available in the official website it's only a fw upgrade.
It's a partial image, you can't use to overwrite completely your firmware.

maximser, 2020-09-26:
you are right, im using y23 from this repo. Ok, thanks for info, so if i get global version of this camera(yi home 1080p) and somehow (current dont know) dump the firmware, would it be possible to flash it?

roleoroleo, 2020-09-26:
I think that there are no differences between a CN camera and a US camera (at firmware level).
The suffix is the same, the upgrade repo is the same, the same hack file works...
Probably when the cam starts, read the serial number and change the beahvior if the cam is US or CN.

closed 2020-11-21 by roleoroleo