roleoroleo/yi-hack-MStar #247: NTP function not working

Dehler2, 2020-09-30:
Hi,

I'm running version 0.3.6 on a cam within a closed network created by a Dahua NVR. The NVR has an internet connection and hosts the NTP server, but every camera within the system can only connect to the NVR.

The NVR seems to provide a proper NTP sync, but my Yi Cam won't sync the time. Every time I do a restart the time is set back to 18 April 2020 ~the time I started with this cam.
The NTP server's IP is set and NTPD is switched on.

Is this a bug or am I missing something?

roleoroleo, 2020-09-30:
Please share your config.

Dehler2, 2020-09-30:
please see the configuration files attached. Only the password has been changed for obvious reasons.

[Yi schuur config no-pass.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5306327/Yi.schuur.config.no-pass.zip)

roleoroleo, 2020-09-30:
I will check it.

SebLz, 2020-10-23:
Hi, having similar issue (0.3.6).
The camera is on an isolated sublan with no internet access but can access the openwrt ntp server, however it won't sync time.
I tried to ssh into the yi cam and typed "ntpd -p <my router ip>" but that does nothing either and time reamains blocked to april 17th, 2020.

roleoroleo, 2020-10-25:
I installed a ntp server locally (same subnet as the cam) and I have no problems:
```
login as: root
root@192.168.1.242's password:

                   _     _           _
               _ _|_|___| |_ ___ ___| |_
              | | | |___|   | .'|  _| '_|
              |_  |_|   |_|_|__,|___|_,_|
              |___|
 -----------------------------------------------------
  yi-hack for MStar platform - v. 0.3.6
 -----------------------------------------------------

/home/yi-hack # uptime
 22:16:40 up 0 min,  load average: 0.59, 0.14, 0.04
/home/yi-hack # ps | grep ntp
  989 root      0:00 ntpd -p 192.168.1.10
 1206 root      0:00 grep ntp
/home/yi-hack # date
Fri Apr 17 22:16:57 CEST 2020
/home/yi-hack # date
Sun Oct 25 11:39:32 CET 2020
/home/yi-hack # date
Sun Oct 25 11:39:36 CET 2020
/home/yi-hack #
```

roleoroleo, 2020-10-25:
Or running manually:
```
login as: root
root@192.168.1.242's password:

                   _     _           _
               _ _|_|___| |_ ___ ___| |_
              | | | |___|   | .'|  _| '_|
              |_  |_|   |_|_|__,|___|_,_|
              |___|
 -----------------------------------------------------
  yi-hack for MStar platform - v. 0.3.6
 -----------------------------------------------------

/home/yi-hack # killall ntpd
/home/yi-hack # ntpd -d -n -q -p 192.168.1.10
ntpd: sending query to 192.168.1.10
ntpd: reply from 192.168.1.10: offset:-0.000230 delay:0.006387 status:0x24 strat:4 refid:0x01c89fa2 rootdelay:0.037446 reach:0x01
ntpd: sending query to 192.168.1.10
ntpd: reply from 192.168.1.10: offset:-0.000810 delay:0.002949 status:0x24 strat:4 refid:0x01c89fa2 rootdelay:0.037446 reach:0x03
/home/yi-hack #
```

SebLz, 2020-10-25:
Hello @roleoroleo 
I've done some more testing and found out I had some firewall issues leading to the error. I've now fixed them and the ntpd functionality seems to work just fine. Sorry for the earlier misleading info...

closed 2020-11-21 by roleoroleo