roleoroleo/yi-hack-MStar #250: Support of YI Dome 1080p

RestOp, 2020-10-04:
Hello, roleoroleo

Thanks one more time for perfect job!
Would like to know if one of this cams are supported:

old one:
https://www.amazon.es/YI-Vigilancia-nocturna-movimiento-giratorio/dp/B01MR82BSA
[image]

new one:
https://www.amazon.es/YI-Vigilancia-Inalámbrica-Movimiento-Compatible/dp/B07QZRJ98M
[image]

or maybe Xioami
https://www.amazon.es/Camara-Domo-XIAOMI-MIJIA-Smart/dp/B07P12T1G3
[image]

I description of features i found PTZ over onvif. В not you know - i can control PTZ in Homeassistant?

Thanks a lot!

roleoroleo, 2020-10-04:
It's not possible to know if it's supported without model details.
Part number and fw version for example.

I think ptz via onvif is supported by last ha.

RestOp, 2020-10-04:
Thanks, will try to buy!
But normaly PTZ supported with onvif apps?

roleoroleo, 2020-10-04:
I have only yi home cams. So I don't use ptz with yi.
But I'm using the same onvif server in the sonoff project.
And it works correctly with odm, onvifer and other programs.
So, yes, it's supported.

RestOp, 2020-10-07:
I have got this one
https://www.amazon.es/YI-Vigilancia-nocturna-movimiento-giratorio/dp/B01MR82BSA
Seems that it is supported, FW 4.6.0
I would like to make a backup of originl FW. Can i make it also as Allwinner method https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Dump-your-backup-firmware-(SD-card) ?
Or i will not be able to return to original FW if i will flash with hack?

Thanks a lot!

roleoroleo, 2020-10-07:
I didn't find an SD card method with MStar.
Only with a serial connection.

closed 2020-11-27 by roleoroleo