roleoroleo/yi-hack-MStar #251: sftp upload

bepece1, 2020-10-05:
Hi, is it possible sftp upload ???

roleoroleo, 2020-10-05:
No, but scp is possible.

closed 2020-11-21 by roleoroleo

Chekedaudau, 2021-05-29:
Dear roleoroleo, first thank you for your awesome work! Could you please explain how to configure scp for the yicam and is the connection secure? I don't want to upload the images/videos to a server on the internet without a secure connection. Is there any idea to implement an sftp upload? Greetings Thomas

roleoroleo, 2021-06-02:
At the moment it is not in the plan.
But scp command works.
So you could try to edit ftppush.sh and move ftp to scp.

LeidenSpain, 2021-09-16:
Hi,

Sorry to reopen the topic, it was just to know if it was still not planned to include support for sftp.
I have a sftp server at home for everything, and it would be very convenient for me to use it without having to create an ftp server.

I have looked at what you say about scp, but I am not sure how to use it

roleoroleo, 2021-09-17:
sftp is not supported at the moment.
You could try to use scp instead. Probably your sftp server supports also scp protocol.
This is a guide:
https://linuxize.com/post/how-to-use-scp-command-to-securely-transfer-files/

LeidenSpain, 2021-09-17:
thanks for the reply

yesterday I was looking at that website (I did not know the scp or what it was), but what I understood was all through the command line. And I don't know how to use command line with the camera

I want to be able to upload a copy of the videos to my sftp when something is detected, to have a backup

Regards