roleoroleo/yi-hack-MStar #255: FTP upload with the directories?

hetzbh, 2020-10-16:
Hi,

The FTP Upload option is really great for archival purposes.
However, it's copying the files, and without the directory itself (which has the date and time when the video was recorder) the MP4 are useless to find what recorded when.

Is there any way to modify the upload to copy the directories and the files, not just the files? 

Thanks

roleoroleo, 2020-10-16:
See this commit:
https://github.com/roleoroleo/yi-hack-MStar/commit/e941ad06e77cc3a57da772177ae7f2cb25927785

hetzbh, 2020-10-16:
So I guess it will be with the next version?

roleoroleo, 2020-10-17:
Yes. Next version.

radokristof, 2020-11-15:
@roleoroleo the FTP upload seems to work nice even with DIR_TREE and custom path.
However the folders on the camera (or at least in the web interface) are not deleted, only the containing files. Would it be possible to clean up the folders when all files have been uploaded in a folder?

roleoroleo, 2020-11-15:
This should already be the case.

radokristof, 2020-11-15:
I just enabled it now, maybe old directories are not cleaned automatically. I will delete them and see if newly created directories disappear or not.

radokristof, 2020-11-16:
@roleoroleo well, for me it won't delete the folders...

roleoroleo, 2020-11-17:
I will check it.

roleoroleo, 2020-11-17:
https://github.com/roleoroleo/yi-hack-MStar/commit/317c96879471d9adf09f023424d2127b76fccdc6

roleoroleo, 2020-12-04:
Released in 0.3.9.

closed 2020-12-19 by roleoroleo