roleoroleo/yi-hack-MStar #257: brick issue - Yi Dome 1080P (BFUS)

hetzbh, 2020-10-16:
I did something stupid and I tried copying a new busybox over the old busybox (looks like the GDrive that you published needs another version of busybox as the current version doesn't have the "stat" command).

Of course, I got out of space and since it removed the busybox binary, the camera started to have a purple/blue flashing led. I tried to reset, tried to put the image files on the micro SD and reboot the camera - nothing. Yellow LED, doesn't do anything, reset button does not help.

Any suggestion how to unbrick this cam?

hetzbh, 2020-10-16:
Ok, I have another (same) camera which works and have the 3.0.6 firmware with the official 4.6.0 firmware
I can use DD to create dumps of the partitions, but how do I know which one to rename, for example: u-boot_h20m kernel_h20m?

roleoroleo, 2020-10-17:
On this platform, when you do a fw upgrade, the whole partition is flashed.
Both sys (rootfs) and home.
So the busybox binary is overwritten during the fw update.
And the flash operation, is made from u-boot before the os loading.
Try to update again with a previous version (0.3.5).

hetzbh, 2020-10-17:
You were right! downgrading 0.3.5 worked and now the camera came back to life. I'll make a video about that.

2 questions, if you don't mind:

* Do you plan to merge the Google upload to the next version (I think you'll need to upgrade or recompile busybox to enable the stat command)
* It seems that many things cannot be added due to space limitation. Did you think to add a loop back support and create/download (when the user chooses) a special image that will be installed on the Mirco SD and mounted? that way, if the user removes the MicroSD, the device would reboot and the additional functionality would be missing, but can be re-added again.

roleoroleo, 2020-10-17:
>     * Do you plan to merge the Google upload to the next version? (I think you'll need to upgrade or recompile busybox to enable the stat command)

No, there is no place in the flash partitions. Will be available as a "sd plugin".

>     * It seems that many things cannot be added due to space limitation. Did you think to add a loop back device support and create/download (when the user chooses) a special image that will be installed on the Mirco SD and mounted? that way, if the user removes the MicroSD, the device would reboot and the additional functionality would be missing, but can be re-added again.

I can't recompile the kernel. We don't have kernel sources and config.

hetzbh, 2020-10-17:
SD Plugin. Is this something you plan for the future or are there other SD plugins? I would love to test those...

roleoroleo, 2020-10-17:
I mean that you can install it to the sd card as you can do today.

hetzbh, 2020-10-17:
ok, but from my testing yesterday, it requires **stat** which is not available on the installed version of busybox.

roleoroleo, 2020-10-17:
What repo are you using?
I'm using it without problems.

closed 2020-11-21 by roleoroleo