roleoroleo/yi-hack-MStar #259: Onvif on hikvision

guyguy1989, 2020-10-17:
I try to add camera on hikvision nvr but the chanel is 0 and on nvr its start 1-16
[image]

The rstp work good but i want to add camera onvif beacuse i want it recorder on my nvr

Thanks
Guy

roleoroleo, 2020-10-17:
It shouldn't matter the channel number.
ch0 and ch1 are only a text string.
Set channel 1 and check if it uses the first profile.

The only difference between onvif and rtsp is the ptz (if the cam is a dome).
When a client want to read the stream with onvif, onvif uses rtsp.
So, if you want to record the stream of your camera you can use rtsp directly or via onvif, it's the same.

guyguy1989, 2020-10-26:
I can see the camera via rtsp but I want the camera to record on the hikvision recording device and not on the memory card that is inside the camera how can this be done?

roleoroleo, 2020-10-26:
Add the camera as rtsp.

guyguy1989, 2020-10-26:
> ‫ב-26 באוק׳ 2020, בשעה 9:36, ‏‏roleo ‏<notifications@github.com> כתב/ה:‬
>
>
> Add the camera as rtsp.

I add but i can view on hikvision record

guyguy1989, 2020-11-05:
???

roleoroleo, 2020-11-05:
If rtsp works with vlc it must work with your dvr.

guyguy1989, 2020-11-05:
> ‫ב-5 בנוב׳ 2020, בשעה 9:29, ‏‏roleo ‏<notifications@github.com> כתב/ה:‬
>
>
> If rtsp works with vlc it must work with your dvr.

I can view record on nvr?

myromeo, 2020-11-10:
I’ve had some problems with my cameras on annke (hikvision clone) NVR and they seem to have been resolved by changing the transfer protocol to UDP. Maybe worth a try.

roleoroleo, 2020-12-19:
I tested yesterday the last version with an Hikvision dvr and it works.
I close this issue.

closed 2020-12-19 by roleoroleo

guyguy1989, 2021-09-07:
Hey again now its work for me, but why on playback its show me always time and not when motion detection?