roleoroleo/yi-hack-MStar #260: Turn ON/OFF

rul3zz, 2020-10-18:
Hello and tnx for this amazing project.
I'm using the camera with Home Assistant, i use without problems PTZ with some rest commands found here.
I wish I could turn off/on the camera via Homa Assistant, can u help me?

Thanks

hetzbh, 2020-10-18:
This project does not support Home assistant. You'll need to contact Yi support for that.

bonuzzz, 2020-10-18:
@hetzbh everything is possible
@rul3zz check this link and use with shell_command in ha (with curl)
https://github.com/roleoroleo/yi-hack-MStar/wiki/Web-services-description

hetzbh, 2020-10-18:
Oh, you're right, i apologize.

rul3zz, 2020-10-20:
> @rul3zz check this link and use with shell_command in ha (with curl)
> https://github.com/roleoroleo/yi-hack-MStar/wiki/Web-services-description

that's exactly what i was looking for .. thank you very much!!

closed 2020-10-20 by rul3zz