roleoroleo/yi-hack-MStar #261: Backup video with rsync

hetzbh, 2020-10-18:
Hi

I've created a page with instructions how to use rsync to backup all/part of the videos in the camera, where to get rsync etc.

The page is located [here](https://github.com/hetzbh/yi-cams-backup), with credirs to this project authors.

@roleoroleo -- Could you add a link to this page in your README.md file please?

bonuzzz, 2020-10-18:
@hetzbh you could run rsync in remote machine without copying binary to camera and it is preferable to run rsync on more powerful machine

hetzbh, 2020-10-18:
You can't, it fails immediately because it's looking for rsync binaries in both ways.

roleoroleo, 2020-10-18:
I had already compiled rsync, if you need it.
[rsync.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5397774/rsync.gz)

roleoroleo, 2020-10-23:
Link added to README.

closed 2020-10-23 by roleoroleo