roleoroleo/yi-hack-MStar #262: [Question] Support for YI Dome X

adicrescenzo, 2020-10-20:
Hi all,

I bought the model YI Dome X on [amazon](https://www.amazon.it/gp/product/B07QZRJ98M/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1).

Those are the values reported above the box of the camera: 

DID: BFUSY02....
SN: BY30EU....
Model: YYS.3017

I tried to flash the firmware y203c_0.3.6 but nothing happened to the camera, the default firmware is still working.
The firmware version in the Yi Home app is 4.0.0.0F_201909251646

I thought that the DID value is the one to identify the camera model, is that correct? Is this camera supported for the hack? Many thanks in advance.

roleoroleo, 2020-10-20:
DID is not enough.
Based on fw version your cam is not supported.

adicrescenzo, 2020-10-21:
> 
> 
> DID is not enough.
> Based on fw version your cam is not supported.

Is the camera model correct for this hack? Are there any chance for the future to support this firmware version?
Thanks

roleoroleo, 2020-10-21:
I don't know.
To check if it's possible to create an hack we need to dump the memory and analyze it.

adicrescenzo, 2020-10-21:
How can I make it? I would really like to contribute

roleoroleo, 2020-10-21:
See the wiki.

adicrescenzo, 2020-10-22:
Done! Can you give some hints to find the 3 uart pads?

I took some pictures of the circuite board:

Back
[image]

Front
[image]

Front from another angle
[image]

roleoroleo, 2020-10-23:
The chip is MStar but I can't see uart pads.

Roihu, 2020-10-24:
UART pads (circled red) can be found under the speaker module, which is attached to the PCB with two screws (circled green). Unfortunately I do not myself have access to an UART adapter right now, so I cannot dump firmware yet.
[image]

adicrescenzo, 2020-10-24:
> UART pads (circled red) can be found under the speaker module, which is attached to the PCB with two screws (circled green). Will post the dump soon if successful.
> [image]

That’s a great news! Thank you very much for the hint. Tomorrow I’ll disassemble the speaker and I’ll try the make a memory dump with the uart interface.

Roihu, 2020-10-24:
I managed to find my own UART adapter and have started the dump, the process seems to be extremely slow, (rootfs seems like it will take around 2 hours, which would mean home will take around 12 hours as it is six times larger). I suppose increasing the --step argument in the dump script would speed this up, but is there a risk of getting corrupted data then? The code for the dump script suggests a default step size of 256, while the wiki says 64 (which I'm using now)

roleoroleo, 2020-10-24:
The uart you are using hasn't control signals (RTS and CTS).
If you use big blocks data will be corrupted.

Roihu, 2020-10-24:
> The uart you are using hasn't control signals (RTS and CTS).
> If you use big blocks data will be corrupted.

I see, thank you. Guess I'll just leave it running overnight. 
Attached is the swapped sys_y30 file. I will update this message to include the home_y30 file tomorrow once the dump is finished. 
[sys_y30.swapped.img.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5433429/sys_y30.swapped.img.zip)

TheTimeWalker, 2020-11-11:
Any news about this? I can spend a beer as an incentive if this can be hacked

roleoroleo, 2020-11-12:
Please try this:
[y30_0.3.8.tar.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5532237/y30_0.3.8.tar.gz)

But make a backup copy before you try.

Roihu, 2020-11-14:
Awesome, flashed fine and at least for now I didn't find anything that _didn't_ work. ONVIF (including PTZ), RTSP, SSH, Telnet, all seems to work. Thank you!

roleoroleo, 2020-11-14:
Great!

TheTimeWalker, 2020-11-14:
Keeping my promise, a beer here costs 8 CHF, so I took that as the donation price

[image]

roleoroleo, 2020-11-14:
Thank you for your donation!

adicrescenzo, 2020-11-14:
> Awesome, flashed fine and at least for now I didn't find anything that _didn't_ work. ONVIF (including PTZ), RTSP, SSH, Telnet, all seems to work. Thank you!

Great news! Did you flash it copying the files on the sdcard?

wasserpanther, 2020-11-14:
Flashed by extracting the tar.gz provided onto the sd card. Got all Camera functions working (PTZ, RTSP, etc.).
One small issue: It seems that the camera looses the WiFi credentials after reboot, I used the "save config" option to see if they got saved in some of the config files but i could not find any information. Am I missing something there or could this be related to bug #66 ?
Thanks for your work, donation is on its way :+1:

roleoroleo, 2020-11-14:
It's a very old problem that affects all platforms, but only a few users.
I have never been able to solve it

The "Save config" doesn't save wifi informations.
You can save them only through the yi app.

Thank you for your donation!

WestLight-dev, 2020-11-16:
Hi all,

one question: are home_y30 and home_y30m two different cameras? Or can I rename the file home_y30 to homey3m?

Thank you!

roleoroleo, 2020-11-16:
home_y30m and home_y30 are related to the same cam, but with different file formats.
home_y30m is a file created to upgrade a file system, it's compressed bz2 or 7z and crypted and it's processed by a linux script.
home_y30 is a file containing a file system image created to overwrite completely a partition, it's "compressed" squash or jffs2 and it's processed by u-boot.

WestLight-dev, 2020-11-16:
I copied the two files (home_y30 and sys_y30) to the memory card but after rebooting the camera it did not work :-(

adicrescenzo, 2020-11-16:
Hi guys, I tried to make a copy of the firmware with the UART adapter but when I type the commands listed in the wiki like "sf probe", nothing appends. I've used a mac book pro with 115200 ad baud rate. Do you have any suggestions? 

Flashing the new firmware I had no problems at all, everything works as expected. Great work @roleoroleo, my donation is on its way too.

Last question: is it possibile to start the recording with a web service invocation?

roleoroleo, 2020-11-16:
> Hi guys, I tried to make a copy of the firmware with the UART adapter but when I type the commands listed in the wiki like "sf probe", nothing appends. I've used a mac book pro with 115200 ad baud rate. Do you have any suggestions?

No suggestions, try to continue.

> Flashing the new firmware I had no problems at all, everything works as expected. Great work @roleoroleo, my donation is on its way too.

Thank you for your donation.

> Last question: is it possibile to start the recording with a web service invocation?

Like after a motion detection event?

adicrescenzo, 2020-11-16:
> 
> 
> > Hi guys, I tried to make a copy of the firmware with the UART adapter but when I type the commands listed in the wiki like "sf probe", nothing appends. I've used a mac book pro with 115200 ad baud rate. Do you have any suggestions?
> 
> No suggestions, try to continue.
> 
Ok I'll do it. If the command takes effect is there any feedback from the camera?
In the guide published on the wiki there is a point "Stop the boot at the U-Boot prompt (should be "MStar #") pressing "Enter"."
Even if I press enter the boot process continues, is it normal? 

> > Flashing the new firmware I had no problems at all, everything works as expected. Great work @roleoroleo, my donation is on its way too.
> 
> Thank you for your donation.
> 
> > Last question: is it possibile to start the recording with a web service invocation?
> 
> Like after a motion detection event?

No, I would like to start the recording manually invoking a web service. In the same way we can move the camera left or right.
Thanks.

roleoroleo, 2020-11-16:
> In the guide published on the wiki there is a point "Stop the boot at the U-Boot prompt (should be "MStar #") pressing "Enter"."
> Even if I press enter the boot process continues, is it normal?

No, check if the TX pin is connected correctly.

> No, I would like to start the recording manually invoking a web service. In the same way we can move the camera left or right.
> Thanks.

Currently is not possible to start a video recording but probably I could try to write a new ws that send a motion detection event to the system. The yi software should start a new record.

roleoroleo, 2020-11-16:
https://github.com/roleoroleo/yi-hack-MStar/commit/f4504af6369bba8085353069dc9e95da50c665b4

Try these:
[ipc_cmd.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5549039/ipc_cmd.gz)
[record.sh.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5549040/record.sh.gz)

adicrescenzo, 2020-11-17:
> [f4504af](https://github.com/roleoroleo/yi-hack-MStar/commit/f4504af6369bba8085353069dc9e95da50c665b4)
> 
> Try these:
> [ipc_cmd.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5549039/ipc_cmd.gz)
> [record.sh.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5549040/record.sh.gz)
> 
> The url is http://IP_CAM:8080/cgi-bin/record.sh?time=30

Great it works!! I've just see a little problem, the invocation to the web service is synchronous and the http call doesn't terminate until the time parameter is reached (e.g 30 or 60 seconds). Is there a way to make this call asynchronous?

Thanks

roleoroleo, 2020-11-17:
Yes, I could fork the process.

WestLight-dev, 2020-11-17:
I need your help! Please! I have a YI Dome X (6FUS) with 4.0.0.0 firmware ...I have tried everything, but nothing helped. I cannot overwrite the firmware.

roleoroleo, 2020-11-17:
Does the camera work normally when you remove the sd card?

WestLight-dev, 2020-11-17:
@roleoroleo 
Yes, it does. I have two other cameras (YI Outdoor 1080p) ... both of them worked immediately.

WestLight-dev, 2020-11-17:
After switching on the camera (with sd card into) the yellow light goes on ... for about 20 seconds. Then the camera goes to the right and back again. Afterwards the light will glow blue.

roleoroleo, 2020-11-17:
Try to change the sd card.

WestLight-dev, 2020-11-17:
I tried with 2GB, 16GB and 64GB ... FAT32 formatted ... very strange :-(

roleoroleo, 2020-11-17:
What the firmware versione?
I need all digits.

adicrescenzo, 2020-11-17:
> 
> 
> @adicrescenzo
> [922affb](https://github.com/roleoroleo/yi-hack-MStar/commit/922affb375d1420db6f07f7dc2577365cf4cca80)

Thank you very much! How can I get the new compiled firmware? I'll try to start the development myself but first of all I need to understand how it works :)

roleoroleo, 2020-11-17:
If you don't want to wait the next release, try to build it yourself.
See the wiki.

thoemmes86, 2020-11-18:
Good Morning,

and many many thanks for offering the support for this camera.
I was waiting a very long time to get this camera supported. So the offering of a beer is for sure.
Both Dome X cameras where able to install the firmware and work like a charm.

BUT:
At least I found one Bug which is only available on one of those identical Dome X Devices.
For ONLY ONE of them, the 'High res snapshot' is not working while the 'Low res snapshot' and all other functionalities are.

The configurations of both cameras are identical, I tried a reset of the camera, but nothing solved the issue.
You may have an idea how to solve?

Kind Regards

WestLight-dev, 2020-11-18:
@roleoroleo : current firmware is **4.0.0.0F_201909251646**

roleoroleo, 2020-11-18:
@thoemmes86 
Is the yi fw version the same?

@WestLight-dev 
The fw versione is ok, it's supported.
Check please if sd is correctly detected from the cam (with yi app).
Are you able to watch the videos recorded from the cam?

thoemmes86, 2020-11-18:
@roleoroleo 
Yes it is exactly the same.

roleoroleo, 2020-11-18:
> @roleoroleo 
> Yes it is exactly the same.

Does high res stream work?

WestLight-dev, 2020-11-18:
sd-card is detected correctly. I can see the logs and videos on the sd-card.

WestLight-dev, 2020-11-18:
[image]

thoemmes86, 2020-11-18:
> > @roleoroleo
> > Yes it is exactly the same.
> 
> Does high res stream work?

Yes everything else is working as expected. (As far as I have tested)

roleoroleo, 2020-11-18:
@thoemmes86 
Try to run the command manually.
`imggrabber -r HIGH -w -d > /tmp/sd/snap.jpg`
and post the output.

thoemmes86, 2020-11-18:
imggrabber -r HIGH -w -d > /tmp/sd/snap.jpg
watermark on
debug on
Resolution: 1080
vaddr: 0xb4745000 - paddr: 0x23570000 - size: 368640
copy buffer: len 368640
copy again buffer: len 368640
Segmentation fault

thoemmes86, 2020-11-18:
So I have identified that there is also an issue with the high res stream which was not that obvious.
RTSP Stream is configured as HIGH.
It is possible to open the RTSP stream, but you only get a low resolution stream.

roleoroleo, 2020-11-18:
It seems the same problem I already encountered on allwinner platform.
Probably it's a different model.
Please make a backup copy of your firmware using dd and share it with me.

thoemmes86, 2020-11-19:
May I'm outing myself as a noob, but currently I was not able to do a DD backup via SSH and storing it on the internal SD.
Do you have some commands I can use for it.

roleoroleo, 2020-11-19:
Login with ssh and run the following commands:
https://github.com/roleoroleo/yi-hack-MStar/issues/223#issuecomment-722469856

Repeat for each mtdblock.

thoemmes86, 2020-11-19:
That is what I tried.

 # cat /tmp/sd/backup/mtd.txt
dev:    size   erasesize  name
mtd0: 00050000 00010000 "BOOT"
mtd1: 00180000 00010000 "KERNEL"
mtd2: 001e0000 00010000 "ROOTFS"
mtd3: 00c30000 00010000 "HOME"
mtd4: 00010000 00010000 "vd1"
mtd5: 00010000 00010000 "conf"
/ # dd if=/dev/mtd0 of=/tmp/sd/backup/mtd0.bin
dd: can't open '/dev/mtd0': No such file or directory
/ # dd if=/dev/mtdblock0 of=/tmp/sd/backup/mtdblock0.bin
dd: can't open '/dev/mtdblock0': No such file or directory

roleoroleo, 2020-11-19:
Try:

```
cat /proc/mtd > /tmp/sd/backup/mtd.txt

dd if=/dev/mtd/mtd0 of=/tmp/sd/backup/mtdblock0.bin
dd if=/dev/mtd/mtd1 of=/tmp/sd/backup/mtdblock1.bin
dd if=/dev/mtd/mtd2 of=/tmp/sd/backup/mtdblock2.bin
...
```

thoemmes86, 2020-11-19:
Worked, thanks.
[backup.zip](https://github.com/roleoroleo/yi-hack-MStar/files/5565301/backup.zip)

roleoroleo, 2020-11-19:
Your hack is ok. I don't understand.

thoemmes86, 2020-11-19:
Hmmm... do you think may a camera hard reset could help or will this lead to potential issues with the hack?

thoemmes86, 2020-11-19:
That's it... a reset with the reset Button at the bottom and a re-pair with the app solved the issue.
Thanks for your time. Could you confirm the deletion of the backup?
As I don't know how good the encryption of yi is... :-D

[24 more replies]

closed 2020-12-19 by roleoroleo