roleoroleo/yi-hack-MStar #263: Not flashing

procrastinate007, 2020-10-21:
Hi, 

I have the Yi Home Camera 3 and I've tried flashing the bios by downloading the files onto a 16GB fat32 microSD card, but to no avail.... should I exchange the cams for the 1080p home version? Is the home 3 not supported? How can I make this work?

Thank you very much for any input!
Appreciated.

roleoroleo, 2020-10-22:
Is your cam in the list of supported cameras?

procrastinate007, 2020-10-22:
I’d say so because it starts with 9FUS..

roleoroleo, 2020-10-22:
You need fw version.

procrastinate007, 2020-10-22:
The app says that my current fw version is 8.3.0.0A_201912271707 --- does that mean it isn't compatible? 

Not quite sure how it works... I'll take any recommendations!

roleoroleo, 2020-10-23:
Try this:
https://github.com/roleoroleo/yi-hack-Allwinner/

procrastinate007, 2020-10-23:
I COULD KISS YOU!!! LOL THANK YOU!!!
Worked like a Charm , thumbs up all the way bro :D hahaha

(so happy rn)

closed 2020-10-23 by roleoroleo