roleoroleo/yi-hack-MStar #268: ONVIF loss

hermitcrabslab, 2020-10-30:
Hi,

I have a very strange bug with my two BFUS(y203c) firmware 0.3.6 using ONVIF on my NVR, after some unpredictable time the signal gets loss and both camera "looks" offline for the NVR.
Tried ONVIF Device manager and get this message on both:
`There was no endpoint listening at http://192.168.0.X/ that could accept the message. This is caused by an incorrect address or SOAP action. See InnerException, if present, for more details.`
The camera are present on web access and RTSP stream is available and working(tested on VLC and also on older NVR that does not support ONVIF but can record RTSP stream) and also High/Low res web snapshot works.

I also have one 9FUS(y20ga) firmware 0.1.9 that is not affected by this bug(does lose signal sometimes but gets back in some seconds)
Camera gets back is I reboot them from web.
I cannot make this automate since video is not actually lost, please HELP!

roleoroleo, 2020-10-30:
Please check if onvif_srvd daemon is running.
Maybe it crashed.

hermitcrabslab, 2020-10-30:
> Please check if onvif_srvd daemon is running.
> Maybe it crashed.

I changed the ONVIF port if still does the same I will check for the daemon, thanks.

hermitcrabslab, 2020-10-30:
This is wird...
I enabled ssh on all 3 camera saved and rebooted all.
After a while 9FUS camera lost the stream, tried to ssh and surprise got denied, went on web(working but slow) and ssh was disabled wtf?
Afer the ssh attempt camera got back the stream 0_O
I am lost.

hermitcrabslab, 2020-10-30:
OK one died again, out of ps aux:
`/home/yi-hack # ps aux
PID   USER     TIME  COMMAND
    1 root      0:01 /init
    2 root      0:00 [kthreadd]
    3 root      0:24 [ksoftirqd/0]
    5 root      0:00 [kworker/0:0H]
    6 root      0:00 [kworker/u2:0]
    7 root      0:47 [rcu_preempt]
    8 root      0:00 [rcu_sched]
    9 root      0:00 [rcu_bh]
   10 root      0:00 [watchdog/0]
   11 root      0:00 [khelper]
   12 root      0:00 [writeback]
   13 root      0:00 [crypto]
   14 root      0:00 [bioset]
   15 root      0:00 [kblockd]
   16 root      0:00 [kworker/0:1]
   17 root      0:57 [kswapd0]
   18 root      0:00 [fsnotify_mark]
   31 root      0:00 [SCLDAZA_THREAD]
   32 root      0:00 [VIPDazaTask]
   36 root      0:00 [deferwq]
   39 root      0:00 /bin/ueventd
   40 root      0:01 [jffs2_gcd_mtd3]
   41 root      0:00 [jffs2_gcd_mtd2]
   79 root      0:00 [mmcqd/0]
   89 root      0:00 [cryptodev_queue]
  100 root      0:00 [spi0]
  128 root      0:00 [kworker/0:2]
  132 root      0:00 [cfg80211]
  142 root      0:00 ./log_server
  143 root      0:41 ./dispatch
  172 root      0:14 [RTW_CMD_THREAD]
  540 root      1:09 /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-g
  594 root     55:40 ./rmm
  670 root      0:00 [kworker/u2:1]
  707 root      0:00 udhcpc -i wlan0 -b -s /home/app/script/default.script -x hostname:YI-LIVING
  808 root      0:06 dropbear -R
  875 root      0:00 httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
  890 root      0:00 dropbear -R
  895 root      0:00 ipc_multiplexer
  915 root      0:00 ntpd -p ro.pool.ntp.org
  936 root      9:30 h264grabber_h -r high -f
  940 root      7:36 rRTSPServer
  944 root      0:09 {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
  950 root      0:00 -sh
  992 root      0:57 wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --type tdn:NetworkVideoTransmitter -
 2168 root      0:00 ps aux
 2268 root      0:00 [kworker/u2:4]
 2538 root      0:00 [kworker/u2:5]
 3267 root      0:00 [kworker/u2:6]
 3887 root      0:00 [kworker/u2:2]
`

roleoroleo, 2020-10-30:
onvif_srvd is not running.
Probably crashed.
I need to know what request is causing the process to crash.
Are you able to capture network traffic between nvr and cam?

hermitcrabslab, 2020-10-30:
> Are you able to capture network traffic between nvr and cam?
If you tell me how... I am a windows guy...

roleoroleo, 2020-10-31:
Copy this file with filezilla or winscp into /tmp/sd
[tcpdump.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5468657/tcpdump.gz)

Login with putty and run these commands:

```
cd /tmp/sd
gzip -d tcpdump.gz
./tcpdump -i wlan0 -w capture1.pcap
```

Then wait for the next problem.

hermitcrabslab, 2020-10-31:
Looks like latest firmware fixed the issue, saw that onwif_srvd was updated, both BFUS flashed and working with no drop since morning.
If stable I will also flash the 9FUS that is mounted outside, not easy to access, and has no sd.
If still bug I will do the tcp dump and upload, thanks.

hermitcrabslab, 2020-11-01:
After 24h still working, I will close the issue, thanks.

closed 2020-11-01 by hermitcrabslab