roleoroleo/yi-hack-MStar #270: Motion detection image event

SebLz, 2020-11-01:
Hello,
First of all, I'd like to thanks you for this amazing work done with this firmware.
I'm a happy user of yi-hack-mstar 0.3.7 on 2 yi dome 1080p cameras (BFUSY).
I've recently tried to configure the cameras with homeassistant following the wiki:
[](https://github.com/roleoroleo/yi-hack-MStar/wiki/Home-Assistant-integration)

Everything seems to work fine except for the motion_detection_image event. It's like this event is never fired although all other topics are well present and working. Could this be a bug ? Here are my config files: https://1drv.ms/u/s!AsjIE9TOb08FvDgSAZwGY5HiTLIV?e=LC9fp1

roleoroleo, 2020-11-01:
What do you use to test the message?

SebLz, 2020-11-01:
I use MQTT Explorer as well as the lovelace UI and none of them show the motion_detection_image topic

roleoroleo, 2020-11-01:
Sorry.
I didn't commit the update of the submodule.
I need to release a new version.

SebLz, 2020-11-01:
Thanks for the quick feedback! I'll test again and report the result whenever the new release is out.

roleoroleo, 2020-11-01:
If you want to overwrite the binary...
[mqttv4.gz](https://github.com/roleoroleo/yi-hack-MStar/files/5471663/mqttv4.gz)

SebLz, 2020-11-01:
Just tried it (renamed old one to .old, extracted the new one in "/home/yi-hack/bin" and rebooted) but with this new version topic "status" stays "offline" and I also have no motion_detection_image topic

roleoroleo, 2020-11-02:
I tested it overwriting the binary.
Status and image are working.
Try to run it manually anch check if there are errors, maybe the configuration is not aligned.

SebLz, 2020-11-02:
Indeed, my bad I forgot to "chmod +x" the new binary ... it's now working here too :)!

closed 2020-11-21 by roleoroleo