roleoroleo/yi-hack-MStar #272: Can`t flash 9FUS Fw4.5.0.0C_201910080934

Ramon-Balaguer, 2020-11-05:
Can`t flash with yi-hack-MStar Firmware.
16 GB SD-Card FAT32 formated.
Both Files, copied to SD-Card.

$ nmap  -Pn 172.17.2.89
Starting Nmap 7.70 ( https://nmap.org ) at 2020-11-05 14:46 GMT
Nmap scan report for 172.17.2.89
Host is up (0.014s latency).
Not shown: 994 closed ports
PORT     STATE SERVICE
21/tcp   open  ftp
22/tcp   open  ssh
23/tcp   open  telnet
80/tcp   open  http
554/tcp  open  rtsp
8080/tcp open  http-proxy

Nmap done: 1 IP address (1 host up) scanned in 1.60 seconds

but when I send a curl:
$ curl 172.17.2.89:8080                                 <HTML><HEAD><TITLE>404 Not Found</TITLE></HEAD>
<BODY><H1>404 Not Found</H1>
The requested URL was not found
</BODY></HTML>

Ramon-Balaguer, 2020-11-05:
In file file.txt i see that all files have a gz extension:

/home/yi-hack/www/images:
total 34
-rw-r--r--    1 root     0             8565 Sep 17 20:01 loading.gif.gz
-rw-r--r--    1 root     0             3269 Sep 17 20:01 logo.png.gz
-rw-r--r--    1 root     0             2946 Sep 17 20:01 icon-hd.png.gz
-rw-r--r--    1 root     0             2758 Sep 17 20:01 touch-icon-iphone-6-plus.png.gz
-rw-r--r--    1 root     0             2520 Sep 17 20:01 touch-icon-ipad-pro.png.gz
-rw-r--r--    1 root     0             2347 Sep 17 20:01 touch-icon-ipad-retina.png.gz
-rw-r--r--    1 root     0             1958 Sep 17 20:01 icon.png.gz
-rw-r--r--    1 root     0             1892 Sep 17 20:01 touch-icon-iphone-retina.png.gz
-rw-r--r--    1 root     0             1224 Sep 17 20:01 touch-icon-ipad.png.gz
-rw-r--r--    1 root     0             1089 Sep 17 20:01 yi-hack-header.png.gz
-rw-r--r--    1 root     0              984 Sep 17 20:01 touch-icon-iphone.png.gz
-rw-r--r--    1 root     0              907 Sep 17 20:01 favicon.png.gz
-rw-r--r--    1 root     0              379 Sep 17 20:01 arrow-down-bold-box-outline.png.gz
-rw-r--r--    1 root     0              374 Sep 17 20:01 arrow-up-bold-box-outline.png.gz
-rw-r--r--    1 root     0              369 Sep 17 20:01 arrow-left-bold-box-outline.png.gz
-rw-r--r--    1 root     0              329 Sep 17 20:01 arrow-right-bold-box-outline.png.gz
drwxr-xr-x    2 root     0                0 Sep 17 20:01 .
drwxr-xr-x    7 root     0                0 Sep 17 20:01 ..

/home/yi-hack/www:
total 2
-rw-r--r--    1 root     0              723 Sep 17 20:01 index.html.gz

Ramon-Balaguer, 2020-11-05:
If i try download gz...
curl http://172.17.2.87:8080/index.html.gz --output test.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   723  100   723    0     0   8123      0 --:--:-- --:--:-- --:--:--  8123

 $ gunzip test.gz

 cat test
<!DOCTYPE html>
<html lang="en">
<head>
<meta charset="utf-8">
<title>Yi Cam Config</title>
<meta name="author" content="gaggi, TheCrypt0, ramundomario, roleoroleo">
<meta name="viewport" content="width=device-width, initial-scale=1">
<link rel="stylesheet" href="css/all.css">
<link rel="icon" type="image/png" href="images/favicon.png">
<link rel="apple-touch-icon" sizes="57x57" href="image/touch-icon-iphone.png">
<link rel="apple-touch-icon" sizes="76x76" href="image/touch-icon-ipad.png">
<link rel="apple-touch-icon" sizes="120x120" href="image/touch-icon-iphone-retin                    a.png">
<link rel="apple-touch-icon" sizes="152x152" href="image/touch-icon-ipad-retina.                    png">
<link rel="apple-touch-icon" sizes="167x167" href="image/touch-icon-ipad-pro.png                    ">
<link rel="apple-touch-icon" sizes="180x180" href="image/touch-icon-iphone-6-plu                    s.png">
<link rel="icon" sizes="192x192" href="image/icon-hd.png">
<link rel="icon" sizes="128x128" href="image/icon.png">
<script src="js/jquery.min.js"> </script><script src="cgi-bin/hostname.js"> </sc                    ript><script src="js/all.js"> </script>
</head>
<body>          <nav class="nav-show"><div class="container">                  <                    ul>
<li>                        [image]                    </li>                                        <li>                        <a href="/?page=configurations">                    Configurations</a>                    </li>                    <li>                                            <a href="/?page=mqtt">MQTT</a>                    </li>                                        <li>                        <a href="/?page=camera_settings">Camera Settin                    gs</a>                    </li>                    <li>                        <                    a href="/?page=eventsdir">Motion Events</a>                    </li>                                        <li>                        <a href="/?page=snapshot">Snapshot</a>                                        </li>                    <li>                        <a href="/?pa                    ge=ptz">PTZ</a>                    </li>                    <li class="nav-elem-                    right">                        <a href="/?page=maintenance">Maintenance</a>                                        </li>                    <li class="icon">                                            <a class="nav-toggler ios-onclick-fix">&#9776;</a>                    </li>   <                    /ul>
</div>        </nav><div id="container" class="container">        </div>       <                    /body>
</html>

roleoroleo, 2020-11-05:
Don't worry for .gz extension, your browser will decompress and open the pages.
Try with a "normal" browser.

Ramon-Balaguer, 2020-11-05:
[image]

My browser show a error, why? 
I use a Versión Chrome 86.0.4240.183 (Build oficial) (64 bits)

Ramon-Balaguer, 2020-11-05:
I use chrome versión 86.0.4240.183 (Build oficial) (64 bits)

[image]

roleoroleo, 2020-11-05:
Maybe a problem during flash operations?
Try to downgrade to the previous version.

closed 2020-11-06 by Ramon-Balaguer

roleoroleo, 2020-11-06:
Did you solve?