roleoroleo/yi-hack-MStar #274: Yi BFUS 4.6 issue

Howy711, 2020-11-11:
I was able to get the firmware updated with the correct files. However when I try to access the camera control panel using the http://ip-cam:8080 or the camera IP address, I am unable to access the control interface. I tested to see if the ONVIF feed is working and I am able to pull that up using a third party program. 

Does anyone have any ideas on how I could or why I would not be able to access the control panel interface?

roleoroleo, 2020-11-21:
Are you able to access with ssh?

ausshir, 2021-07-24:
This is an old issue, but I have the same camera with the same firmware, and was seeing the same issue!

I can not connect via HTTP, nor SSH. Actually, ping doesn't respond either.

I can still connect with the Yi Home app and confirmed the IP on my router.

_Of note, my wifi network was using 10.2.x.x!_

I switched to another network with 192.168.x.x, set up wifi again and was able to connect to the configuration page.

Is it possible somehow 10.2.x.x networks are not allowed?

roleoroleo, 2021-08-04:
afaik the cam should work with your subnet.
But there is something strange in your test: how can the app work and not ping?

ausshir, 2022-01-04:
Hey sorry for the long time without a response. I ended up going with a different platform so I might not be able to debug further.

However, Ping works with ICMP (different from TCP/UDP) so I have a feeling the listener was just not listening on that subnet either.

closed 2024-05-10 by github-actions[bot]